mantidproject/mantid #29326: Mantid workbench has the wrong icon on nightly

AnthonyLim23, 2020-09-07:
Open workbench (I used the launch script) on nightly. The icon will be the release one

martyngigg, 2020-09-07:
@AnthonyLim23 For the icon, do you mean the `.exe` icon or the desktop shortcut? Could you include a screenshot here?

AnthonyLim23, 2020-09-07:
[image]

and it crashed when I closed it...

martyngigg, 2020-09-07:
Ah okay. So it's the taskbar icon. Thanks.

martyngigg, 2020-09-08:
@AnthonyLim23 I couldn't see in the code where we change this so I downloaded a nightly from June and it appears to be the same:
[image]

Do remember it behaving differently? I think it is probably a good idea to use different icon but if we haven't regressed can we drop the "High Priority" label and see if we get time to look at it and assess risk?

AnthonyLim23, 2020-09-09:
I thought I had seen it show up at nightly at some point. It may have been plot. I will remove the high label

martyngigg, 2020-09-09:
Thanks. I checked MantidPlot too and I couldn't see it. Let's keep this one for a future sprint.

closed 2022-05-01 by stale[bot]