mantidproject/mantid #29328: Qt message in terminal when plotting

gvardany, 2020-09-07:
Plotting a workspace in workbench started to throw this message in the terminal:

`QMainWidget::resizeDocks: all sizes need to be larger than 0`

All the platforms seem to be affected.

gui-co, 2020-09-07:
This seems to come from that line, added to answer #26999:
https://github.com/mantidproject/mantid/blob/e52d80d346bd3988e19fd4c828620d18cd972bbf/qt/applications/workbench/workbench/plotting/figuremanager.py#L241

Removing the line on Ubuntu 18.04 (Qt 5.9.5) does not reintroduce the original bug and avoids the `QMainWidget::resizeDocks: all sizes need to be larger than 0` messages when plotting. However, this needs to be tested on other platforms.

A safer way would be to change the value to 1 has it is what Qt will do after throwing the message:
https://code.qt.io/cgit/qt/qtbase.git/tree/src/widgets/widgets/qdockarealayout.cpp?h=v5.9.5#n3176

martyngigg, 2020-09-07:
@gui-co Thank you for tracking this down. @clayton-tom It looks like the change we made here when fixing the fit browser resizing have caused this. 

Let's set this to 1 in those instances so that we don't reintroduce the bug.

martyngigg, 2020-09-07:
@gui-co Would you be happy to look at this?

clayton-tom, 2020-09-07:
@martyngigg apologies, I thought you were asking me to give it a look! In any case there's a pr up that I think alters the zero parameter in all cases I remember the workaround being used

gui-co, 2020-09-08:
I am a bit late apparently... sorry.

martyngigg, 2020-09-08:
@gui-co No problem. This was my fault for not being clearer with my initial message.

PR #29340 (2020-09-08): Fix resizeDocks 0 value warning
**Description of work.**
Replace some resize calls from a previous workaround to use `1` instead of `0`

**To test:**

Check that plotting in workbench does not throw an error

Fixes #29328. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If ther [... cut]
closed 2020-09-09 by LamarMoore