mantidproject/mantid #29332: Investigate spurious save-project prompt when close quickly after start

gemmaguest, 2020-09-07:
### Expected behavior
You should not get prompted to save a project if there are no changes

### Actual behavior
You get prompted to save the project if you close MantidPlot while the spash screen is still showing. This prompt may also be hidden behind the splash screen so that you can't interact with it. I am also then seeing a segfault when MantidPlot closes.

### Steps to reproduce the behavior
- Start MantidPlot
- While the splash screen is showing, hit the X in the corner to close MantidPlot
- The save-project prompt pops up

### Platforms affected
Seen on Ubuntu with MantidPlot - not yet sure if it affects other platforms or Workbench

gemmaguest, 2020-09-11:
Unassigning myself because I no longer have time to look at this

robertapplin, 2020-09-16:
Tested on windows, but could not replicate this. It might be ubuntu-specific so unassigning this issue

DavidFair, 2020-09-18:
I cannot reproduce this with a Debug or Release build on Windows / Linux

I suspect the timing is extremely specific that Gemma happened to run into. Since it's likely been "broken" for a long time, is very difficult to reproduce and were removing Mantid Plot soon I'm going to close

closed 2020-09-18 by DavidFair