mantidproject/mantid #29339: Unhandled expection in Engineering Diffraction Fitting

StephenSmith25, 2020-09-07:
Found during unscripted testing. In the fitting tab of Engineering diffraction both TOF and dspacing appear as in the workspace selection menu. If you have TOF data plotted and switch this workspace to d-spacing and click the fit button you'll get an unhandled exception, which brings up the error reporter. Ideally, the workspace selection menu should only allow you to pick from the plotted data.
### Steps to reproduce the behavior
 Create some data as per #28393

- Go to fitting and load one of the focussed files.
- Plot the data and open the fit pane
- Add a fit function
- If you changed the selected workspace and press fit you'll get the exception.

### Platforms affected
All

RichardWaiteSTFC, 2020-09-08:
Just to check this doesn't occur if the workspace are all added to the plot and your selecting between them in the fitprop browser - it is when you remove the from the plot using the checkbox in the table?

StephenSmith25, 2020-09-08:
Yep its all fine if the workspaces are plotted. Maybe I had just got the GUI into a bad state, as I was messing around with it for a while. So may be worth seeing if the same happens for you.

RichardWaiteSTFC, 2020-09-08:
No I've come across this error before, thanks for reminding me/catching it. There were so many other things to sort out it wasn't a high priority, but that's not the case now. FYI I think if you put the data back on the plot it works again - maybe you could check this too? It is quite easy to get it into a confused state atm. I haven't decided on the best way to deal with it, maybe you have some ideas? The options as I see it are (1) Close and re-open the fitproperty browser every time the plot updates so when it re-opens you have an up to date list of the workspaces, but this disrupts the user's workflow, you loose the parameters you'd set etc. (2) same as above but close and re-open when the input workspace is removed from plot (3) catch the error and issue a warning (4) fix the code in fitpropertybroswser that crashes - I think in this case keep track of the normalisation.

StephenSmith25, 2020-09-08:
Yep was all good once I had the data back on. Might be a hassle but ideally the plot and combo box should be in sync. Presumably you're using fitpropertybrowser.py? As it seems the method `_get_allowed_spectra` should be handling this for us.

PR #29413 (2020-09-16): Update input workspaces in fitproperty browser of EngDiff UI to stop exception
**Description of work.**
There was an unhandled exception when attempting to fit a workspace that is not plotted - to fix this it is now only possible to select plotted workspaces.

Every time a spectra is added or removed from the axes the fit property browser is updated (by hide/show). If a workspace is removed from the plot then all the spectra associated with that workspace (only one in this case) are removed from the list which populates the drop-down box of input Workspaces. In addition the evaluated fit funct [... cut]
closed 2020-09-16 by martyngigg