mantidproject/mantid #29344: Indirect ISIS Calibration - background and rebinning ranges do not update after changing reflection

robertapplin, 2020-09-08:
### Expected behaviour
The binning range and background width of the resolution should be set automatically. This does not happen if you change the Reflection after entering Input Runs. When changing the reflection, these ranges should probably be updated.

### Actual behaviour
The binning range and background range is given a width of zero, which is not helpful!

### Steps to reproduce the behaviour
1. `Interfaces`->`Indirect`->`Data Reduction`->`ISIS Calibration` tab
2. Enter run numbers `59057-59059`, wait for them to load
3. Check `Create RES File`
4. Change reflection to `004`
5. Click `Run`
6. This message appears in the log:
[image]

### Platforms affected
Tested on windows nightly build

robertapplin, 2020-09-08:
I think this is a low priority bug, so I am not sure if it should be fixed for this release. @SpencerHowells what do you think?

SpencerHowells, 2020-09-08:
@robertapplin Low priority. The reflection should be changed before loading data.

SpencerHowells, 2020-09-09:
@robertapplin More thoughts. The run is different for the 2 reflections so if you change a reflection you should change the run.

closed 2022-02-23 by stale[bot]