mantidproject/mantid #29352: Sliceviewer swapping axes doesn't work

joseph-torsney, 2020-09-08:
### Expected behavior
Clicking the buttons to swap which value is displayed on each axis should swap what is displayed on each axis, and the data should be plotted correctly.

### Actual behavior
This is easier to explain with pictures...
1. This is how the plot looks initially 
[image]

2. After clicking to swap the axes once, the axes are correct but the data is plotted incorrectly, the max value on the `Spectrum` axis is the same as the max value on the `Time of flight` axis (if you drag the plot to the right).
[image]

3. Swapping the axes again swaps the labels but not the actual plot
[image]

4. Swapping the axes again plots the data correctly, but the range on the axes appear incorrect
[image]

5. Swapping the axes again gets us back to the correct state, and to step 1.
[image]

### Steps to reproduce the behavior
See above

### Platforms affected
Tested on windows

Matt-Cumber, 2020-09-08:
There is also a weird visual glitch to these buttons on Mac (at least for me) for example:
**Using the MAR11015 dataset**

All buttons can be pressed at the same time
[image]

Or all off
[image]

Or they are half on
[image]

joseph-torsney, 2020-09-10:
Could be related to #29400

Matt-Cumber, 2020-09-10:
Regarding my previous comment, if you swap the axis by pressing the button once then click off Mantid onto your desktop or another application, then click back on Mantid the visual bugs sort themselves out. Still doesn't look quire right though 😕

clayton-tom, 2020-09-11:
Merging with #29400

closed 2020-09-11 by clayton-tom