mantidproject/mantid #29355: Unhandled TypeError exception plotting from Data table

clayton-tom, 2020-09-08:
I've only found the issue and been able to reproduce on unfocused GEM files, specifically GEM40979 from the Training Course Data. Focused GEM files and other ws2Ds I've tried seem fine.

### Steps to reproduce the behavior
1. Load GEM40979
2. Show Data
3. Select one or more columns
4. Right click, plot bin

### Platforms affected
Discovered on windows, assumed on all

clayton-tom, 2020-09-08:
Also happened with SXD23767

martyngigg, 2020-09-08:
@clayton-tom Could you provide details of the error message that is created. 

I'm not sure I would count this as high priority unless we know it's a regression from last release? If not then I'm happy to keep it on the board but I think we should remove the "High Priority" label as that's only for things that should hold up the release.

martyngigg, 2020-09-11:
Closing as auto close when merging with release-next doesn't work.

PR #29387 (2020-09-10): Fix issue with numpy.where tuple return
**Description of work.**

The `get_bin_indices` function in in `datafunctions.py`  uses `numpy.where` to determine the correct bin indices. However, not accounting for the fact that numpy.where returns a tuple.

**To test:**

Steps to reproduce the behavior
Load GEM40979
Show Data
Select one or more columns
Right click, plot bin (you should successfully plot the bins instead of seeing a type error in workbench)

Fixes #29355. 

*This does not require release notes* 

---

#### Reviewer ####

Please comment on the following ([full description](http://deve [... cut]
closed 2020-09-11 by martyngigg