mantidproject/mantid #29362: System tests pass on Ubuntu 20.04

AndreiSavici, 2020-09-09:
Part of #28650. Make sure all system tests pass on Ubuntu 20.04

AndreiSavici, 2020-09-09:
```
FAILED:
CalibrateRectangularDetector_Test.PG3Calibration
CrystalFieldPythonInterface.CrystalFieldPythonInterface
EnggCalibrationTest.EnginXCalibrateFullThenCalibrateTest
PVPythonTest.PVPythonTest
VesuvioFittingTest.VesuvioFittingWithKFreeTest

98% tests passed, 5 tests failed out of 482 (24 skipped)
```
 * `CalibrateRectangularDetector_Test.PG3Calibration` - data mismatch
 * `CrystalFieldPythonInterface.CrystalFieldPythonInterface` failed because no pycifrw installed
 * `EnggCalibrationTest.EnginXCalibrateFullThenCalibrateTes` - wrong fitting result 2.24!=2.32
 * `PVPythonTest.PVPythonTest` - no paraview installed
 * `VesuvioFittingTest.VesuvioFittingWithKFreeTest` - data mismatch

martyngigg, 2020-11-20:
CrystalFieldPythonInterface was failing because we had no PyCifRW build for 20.04. This is now in the mantid PPA so this should fix that test.

AndreiSavici, 2021-07-26:
@peterfpeterson This is our internal issue https://code.ornl.gov/sns-hfir-scse/spectroscopy/spectroscopy/-/issues/44