mantidproject/mantid #29389: Things spotted testing ISIS SANS Interface in Workbench Nightly

smk78, 2020-09-09:
Umbrella issue whilst testing 5.0.20200908.1743 on W10/x64.

- [ ] Attempting to export the current state of the (reduction) table to csv generates an unhandled exception if no Batch File is specified. Messaging reports:
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\scripts\Interface\ui\sans_isis\sans_data_processor_gui.py", line 455, in _export_table_clicked
    self._call_settings_listeners(lambda listener: listener.on_export_table_clicked())
  File "C:\MantidNightlyInstall\scripts\Interface\ui\sans_isis\sans_data_processor_gui.py", line 230, in _call_settings_listeners
    target(listener)
  File "C:\MantidNightlyInstall\scripts\Interface\ui\sans_isis\sans_data_processor_gui.py", line 455, in <lambda>
    self._call_settings_listeners(lambda listener: listener.on_export_table_clicked())
  File "C:\MantidNightlyInstall\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 120, in on_export_table_clicked
    self._presenter.on_export_table_clicked()
  File "C:\MantidNightlyInstall\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 706, in on_export_table_clicked
    default_filename = self._table_model.batch_file
AttributeError: 'TableModel' object has no attribute 'batch_file'
```

- [ ] Many real-world distances on the different tabs have been converted from mm/cm to m. This is slightly annoying but I can guess the rationale. _However_, the units displayed alongside these boxes have not been updated! This needs fixing quickly or users will be very confused.

  But some distances have not (yet) been updated. For example:
  - [ ] Z offset has no units at all.
  - [ ] I have not checked Monitor 4/5 shifts, or ROI radius.
  - [ ] I have not checked Rcut or H1 & W1 for Slit Resolution.
  - [ ] The radius limits in Settings > Q, Wavelength, Detector Settings are now in m, but the radius limits in Beam Centre are still in mm

- [ ] In Settings > Mask, no information is displayed unless you click Display Mask, even though it must have been read in when the User File was loaded.

gemmaguest, 2020-09-10:
@smk78 We spotted the batch export problem in our own testing and I have a fix for this which should go in today. We also spotted a problem with summing runs which I'm looking at now: #29358

I'll fix the confusion with units too as that is pretty serious! How important is the mask information not being displayed by default? Can this wait till the next release? I'd rather avoid making non-critical changes at this stage for fear of breaking something else...

smk78, 2020-09-10:
Yes, the Display Mask issue is less serious.

gemmaguest, 2020-09-10:
@smk78 I believe that all of the values from the user file are stored internally in m now, which makes sense for programmer sanity, but I don't think it would be difficult to convert back to mm for the view (for the relevant values) if that is preferable?

It looks like the beam centre radius limits are hard coded so have not been changed, so for consistency I will look at storing them in m internally and again can convert to mm for the view.

smk78, 2020-09-10:
@gemmaguest It's an interesting question. I don't have a problem with internal storage in m, indeed, that's probably sensible. But I think we need to be careful about dogma causing usability issues. For example, the Z Offset in the User File I created yesterday is -6 mm. Which is displaying as -0.006 (m). If one needed to change that, let's say to -11 mm, it would be all too easy to enter -0.0011 instead of -0.011. So there may be places in the GUI where mm more sensible to avoid confusion?

This might be a case of let's see what people complain about!

gemmaguest, 2020-09-10:
Yes I think I'm leaning towards keeping mm in the GUI if possible. The beam centre finder results pane is still showing values in mm, too.

We are also displaying values to 19dp in some cases due to machine precision, which is not pretty!

smk78, 2020-09-10:
Agreed, 0.1 microns is probably sufficient precision!

smk78, 2020-09-11:
Richard Heenan comments:

- [ ] 1.
  > What is the “Optimization” group supposed to do?  It seems to have a can, but only appears the first time it is used, unless you delete it, when it comes back for the next reduction. It had a bit more when I did “All” or “Merged” reductions.

  then added:

  > It is showing numerators & denominators, and their division, sometimes for can. These are usually internal workspaces which then get deleted, but there is a flag in mantidplot to keep them visible. But then they ought to be there for sample as well as can?

- [ ]  2. If you do a DET/BOTH reduction you get the LAB and HAB workspaces and, if you have asked for output files, these get written out. If you do a DET/MERGED reduction you get the LAB, HAB and Merged workspaces, but only the Merged workspace is written out. Whilst there is a sort of logic to this, it is mighty irritating if you want to save all three files!

- [ ] The detector dropdown selector offers ALL, but the (old) User File command is BOTH (DET/ALL actually fails). It would be good to standardise on one or other and implement that in the TOML parser.

  >

smk78, 2020-09-11:
More from Richard!

- [ ] 5. Selecting 1.75:2.0:16.5 in the limits wavelength box, with lin steps of 0.125 CRASHES Workbench! (According to the tool tip, putting 1.75:2.0:16.5 ought to work !).
  But the other method  of entry in wavelengths box, 1.75-3,75,3.75-5.75,5.75-7.75, etc ALSO CRASHES Workbench!
  And listing wavelengths with just commas, 1.75,3.75,5.75, etc  stops with an unhelpful error message:
  The provided event slice configuration ['1.75', '3.75', '5.75', '16.5'] cannot be parsed because of 1.75

- [ ] 6. The wavelength step box displays the wrong units when the wavelength range is Lin or RangeLin: it shows Ang^-1 instead of Ang.
[image]

gemmaguest, 2020-09-14:
> 
> More from Richard!
>     * [ ]  5. Selecting 1.75:2.0:16.5 in the limits wavelength box, with lin steps of 0.125 CRASHES Workbench! 

This is a serious regression; we'll get it fixed
 
>     * [ ]  6. The wavelength step box displays the wrong units when the wavelength range is Lin or RangeLin: it shows Ang^-1 instead of Ang. #29442

This has been the same since at least v5.0.0 and so we're not currently prioritising this for this release unless you think it's particularly important. It's probably a straightforward fix but we prefer to avoid making too many changes this late in the release cycle in case it introduces more bugs!
 
>     * [ ]  7. With Log Level = Notice, reducing SANS2D data am getting lots of red messages about "MaskInstrument is deprecated, use MaskDetectors instead". A Windows error message box also pops up. _But_ Workbench continues! #29443
>       If these messages are serious, why does Workbench continue?
>       If they are not serious, are they appropriate at Notice level? And why are they generating a Windows error?

Sorry, this should have been updated to use the new algorithm so you do not see the messages. They are not serious but I would like to get it fixed to avoid worrying/confusing users. Notice level is normally appropriate for deprecation warnings (although it's a bit different here because this comes from internal code). I'm not sure about the Windows error - I've added a note to the linked issue to check this.

gemmaguest, 2020-09-14:
> Richard Heenan comments:
>
>     * [ ]  1.
>
>
>       then added:

I don't know much about these. Do you need/use them? I can investigate but since the behaviour is the same as v5.0.0 I don't think there is anything here that needs addressing for this release? 

>     * [ ]   2. If you do a DET/BOTH reduction you get the LAB and HAB workspaces and, if you have asked for output files, these get written out. If you do a DET/MERGED reduction you get the LAB, HAB and Merged workspaces, but only the Merged workspace is written out. Whilst there is a sort of logic to this, it is mighty irritating if you want to save all three files! What is DET/ALL supposed to do?
> 
>     * [ ]  3. The detector dropdown selector offers ALL, but the (old) User File command is BOTH (DET/ALL actually fails). It would be good to standardise on one or other and implement that in the TOML parser.

Yes it looks like All and Both are the same - I'm not sure what it is in the TOML schema but we can tweak this or update the gui to match, whichever you prefer. This should be a job for the next release, though.

>     * [ ]   4. With auto-generated workspaces names, when you do a MERGED reduction an extra _LAB or _HAB is being appended to the names.

I'm not sure what you mean here - it looks like the optimization workspaces include `_HAB` and `_hab`, or `_main` and `_lab`, which does seem unnecessary, but that is always the case regardless of whether doing a merged reduction. Happy to change those names in the next release - just let us know what you want.

smk78, 2020-09-14:
> Yes it looks like All and Both are the same - I'm not sure what it is in the TOML schema but we can tweak this or update the gui to match, whichever you prefer. This should be a job for the next [...]

I've been discussing these with the SANS Group and we would like to:

- keep the behaviour of DET/BOTH as is
- modify the behaviour of DET/MERGED so that all 3 files are written by default (if output to file/both is selected) but the LAB/HAB files are to be written to a \unmerged subfolder
- do away with DET/ALL

Can we do that?

gemmaguest, 2020-09-15:
> I've been discussing these with the SANS Group and we would like to:
>
>     * keep the behaviour of DET/BOTH as is
>
>     * modify the behaviour of DET/MERGED so that all 3 files are written by default [...]

I can't see why not. I've created #29454 for this and we can discuss it's priority in the next sprint plan.

smk78, 2020-09-16:
eek! I have just found a blocker! This is testing 5.0.20200914.1537.

- [ ] **2D reductions are not working at all!** If only using a User File at the top of the interface, checking the 2D button still generates 1D reductions and output, even though that output is named as _2D_. If you then put a User File in the table the output is only named as _1D_ and is indeed also 1D, even though the 2D button is checked.

DavidFair, 2020-09-16:
> eek! I have just found a blocker! This is testing 5.0.20200914.1537.
> 
>    **2D reductions are not working at all!** 

Sorry that was a pretty major one to let slip. I set one of two flags to be 2D which meant half our reduction was 2D but we used Q1D instead of QXY. #29483 Should fix this one

smk78, 2020-09-16:
- [ ] When you load a User File, the beam centre coordinates are being updated in the Beam Centre tab, _but_ the LAB coords are being written in both the LAB and HAB boxes; ie, the HAB coords are being ignored (or overwritten).

  [image]

  The User File in this case

  [image]

  contains

```
  SET CENTRE 324.645 330.777 5.00 5.00
  SET CENTRE/HAB 321.209 319.683
```

smk78, 2020-09-23:
Testing Nightly 5.0.20200922.1715:

- [ ] If you are not actually displaying the Editor and select File > Generate Recovery Script there is absolutely no indication at all that anything has happened! Might I suggest sending something to Messages?

- [ ] If you have Runs > Save Options > Save Can checked, Process some data, and then display any workspace generated in the folder unsubtracted_can_and_sam _with errors_, your Messages fills up with multiple
```
  Warning: converting a masked element to nan.
```

This _only_ happens if you ask for errors, and only happens on these particular workspaces, not on any other reduced data workspaces!

DavidFair, 2020-09-24:
I've created issues for any outstanding things we didn't pick up - I'm going to close this just so that other developers know we have resolved the important problems for this release

closed 2020-09-24 by DavidFair

smk78, 2020-09-24:
Is the Display Mask issue ticketed?

DavidFair, 2020-09-24:
Yes - Gemma opened it for me (thanks): https://github.com/mantidproject/mantid/issues/29416