mantidproject/mantid #29400: Swapping axis after zooming causes Sliceviewer to crash

tolu28-coder, 2020-09-10:
### Expected behavior
Should be able to swap axis successfully after zooming in on plot.

### Actual behavior
swapping axis after zooming causes an error.

### Steps to reproduce the behavior
1. Open Sliceviewer on any suitable dataset.
2. Zoom in on any area.
3. Try swaping the axis and an error message should pop up.

This was the error produced.
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 171, in dimensions_changed
    self.new_plot()
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 106, in new_plot_matrix
    self.view.data_view.plot_matrix(self.model.get_ws(), distribution=not self.normalization)
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\view.py", line 311, in plot_matrix
    self.draw_plot()
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\view.py", line 333, in draw_plot
    self.colorbar.set_mappable(self.image)
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\colorbar\colorbar.py", line 131, in set_mappable
    self.colorbar = Colorbar(ax=self.ax, mappable=mappable)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 1183, in __init__
    ColorbarBase.__init__(self, ax, **kw)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 460, in __init__
    self.draw_all()
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 493, in draw_all
    self._config_axes(X, Y)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 727, in _config_axes
    self.update_ticks()
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 614, in update_ticks
    ticks, ticklabels, offset_string = self._ticker(locator, formatter)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\colorbar.py", line 850, in _ticker
    b = np.array(locator())
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\ticker.py", line 2439, in __call__
    return self.tick_values(vmin, vmax)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\ticker.py", line 2527, in tick_values
    decades.extend(b ** (np.arange(c_range[0], c_range[1], stride)))
ValueError: arange: cannot compute length

### Platforms affected
Observed on windows but expected to affect all

joseph-torsney, 2020-09-10:
Was unable to reproduce this in windows. Mentioned this in #29352 as they could be related

clayton-tom, 2020-09-11:
I've not been able to reproduce exactly, I didn't get any errors after zooming in a small amount and changing, but when I zoomed in past a certain extent I got a slightly different error on switching axes:
`File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\imageinfowidget.py", line 66, in _on_cursor_at_axesimage
    self._widget.cursorAt(xdata, ydata, arr[i, j])
RuntimeError: indexOfValue - value out of range`
With GEM40979

PR #29467 (2020-09-16): Fix sliceviewer extents to set correctly after axes swap
**Description of work.**
Stop transpose kwarg from being removed, allowing a `true` value to propagate to the code that handles the changing of the image extent.

**To test:**
1. Load a 2D workspace such as MAR1160
2. Open the sliceviewer for this workspace
3. Switch between the axes, trying both sets of x/y buttons, and observe that the image transposes correctly and doesn't throw any errors
4. Repeat after zooming in/out on the image

Fixes #29400 . 

*This does not require release notes* because **it is a regression bugfix**

---

### [... cut]
closed 2020-09-16 by AndreiSavici