mantidproject/mantid #29409: Muon analysis results table

AnthonyLim23, 2020-09-10:
In results table the output no longer has the y errors set in the table columns

https://github.com/mantidproject/mantid/blob/master/scripts/Muon/GUI/Common/results_tab_widget/results_tab_model.py#L276

it might be that the value has changed at
https://github.com/mantidproject/mantid/blob/master/scripts/Muon/GUI/Common/results_tab_widget/results_tab_model.py#L33

AnthonyLim23, 2020-09-11:
Load data
Do a fit
go to results table
Tick a few sample logs
Produce the results table
Check the errors are marked up correctly

Harrietbrown, 2020-09-11:
So this does not look like an issue with muon itself. it's a bug with the displaying of the table itself.

Harrietbrown, 2020-09-16:
Closed by #29434

PR #29434 (2020-09-15): Fix Y error labels in muon analysis results tables
**Description of work.**

The output results table from Muon analysis should label the colombs with `Y0, Y1, Y2...` and `Y0_err, Y1_err, Y2_err...` for the values and errors respectivly. This pr links the two colombs so this label is propperly added to the errors.

**To test:**

1. Open Muon Analysis.
2. Load EMU20918.
3. Perform a fit.
4. Go to the results tab and output the results.
5. check that the colomns in the results table have the right labels linking values with their errors.

Fixes #29409 

*This does not require release notes* beca [... cut]
closed 2020-09-16 by Harrietbrown