mantidproject/mantid #29414: Unhandled exception loading Larmor user files

gemmaguest, 2020-09-11:
**Original reporter:** Rob Dalgliesh, @smk78 

### Expected behavior
The ISIS SANS GUI should support all valid user files that v5.0.0 supported

### Actual behavior
There is an unhandled exception loading at least some (possibly all) Larmor user files

### Steps to reproduce the behavior
Must be tested at ISIS:
- Open the ISIS SANS GUI
- Open this user file \\isis\inst$\ndxlarmor\user\users\Masks\User_Larmor_202A_Changer_r51476.txt
- You get an unhandled exception 

### Notes
- I think this is related to the large user file parsing changes from PR #28513 
- The problem is on [this line](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/sans/state/StateObjects/StateMoveDetectors.py#L279) because `data_info.sample_scatter_run_number` is `None`. I'm not sure if it's valid that this is None so this requires investigation to see why that is the case. It seems odd to be checking for a run number before we have specified any runs, so perhaps this is being called by mistake.

### Platforms affected
All

smk78, 2020-09-11:
Once you incur the exception you are blocked from opening the SANS Interface so you cannot change to a valid User File! The only way I've found to escape the conundrum is to rename the offending User File. This incurs a different error but at least allows the Interface to start.

This behaviour could do with some better error handling.

smk78, 2020-09-11:
What I'm confused by is why if I just start Workbench and load Rob's User File before I do anything else I get the error, but if I do the same thing with a LOQ or ZOOM User File then Workbench is happy. I can't see any thing odd in the LARMOR User File so the error would seem to be LARMOR-specific in the code.

gemmaguest, 2020-09-11:
> Once you incur the exception you are blocked from opening the SANS Interface so you cannot change to a valid User File! The only way I've found to escape the conundrum is to rename the offending [...]

Good point. Most errors that occur on loading are caught and we then leave the user file empty so this does not happen. I'll see if there is some higher level exception handling that we can add to catch unexpected errors like this.

Another workaround for now is to open v5.0.0 and change to a different user file there.

gemmaguest, 2020-09-11:
> What I'm confused by is why if I just start Workbench and load Rob's User File before I do anything else I get the error, but if I do the same thing with a LOQ or ZOOM User File then Workbench is [...]

Yes there is some larmor specific code [here](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/sans/state/StateObjects/StateMoveDetectors.py#L279). It is setting a conversion value based on the run number. I'm not sure what this is about - I vaguely remember we had some problems with some nexus files and had to add a bit of a hack in Mantid to work around it - could it be that?

The problem is that this bit of code is being called while the run number is None - perhaps it is being called too early because I'm not sure why we'd have a run number on loading just the user file anyway! Still investigating...

smk78, 2020-09-11:
> Yes there is some larmor specific code [here](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/sans/state/StateObjects/StateMoveDetectors.py#L279). It is setting a conversion value [...]

I've been in touch with Rob. He says he can't remember for sure but there was a need to convert some log file positions to enable the Z offset to be added to the detector distance to move the detector to the correct position. That makes sense, except every instrument has to do that and this is a LARMOR-specific piece of code for its early runs! So I wonder if it might have been that initially LARMOR was storing its distances in m and not mm like the other instruments???

gemmaguest, 2020-09-11:
It looks like it relates to the beam centre finder, only for pos1, and only for larmor runs after 2217. It looks like it's related to the scale in [SANSBeamCentreFinder](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py#L134) which has some special case code for larmor but does not take the run number into account.

PR #29444 (2020-09-15): Add test for Larmor StateMove throwing if RB is None
**Description of work.**

Adds a test to check for StateMove throwing when the run number is none.
This happens when we add a user file without a run number in the box.

I checked with the SANS team and Larmor uses their x dim in
m, and y in mm for runs after 2217. So we want to use the same divider
to keep the display consistent until they load in run data.

**Report to:** @smk78 

**To test:**

- Open the SANS GUI
- Load the following userfile: 
[USER_Larmor_202A_Changer_r51476.txt](https://github.com/mantidproject/mantid/files/5219147/USE [... cut]
closed 2020-09-16 by LamarMoore