mantidproject/mantid #29415: Units on SANS GUI are wrong

gemmaguest, 2020-09-11:
**Original reporter:** @smk78 

PR #28513 changed some of the units of values on the SANS GUI but the labels have not been updated to match. We should ideally change units back to their originals (mm rather than m) but could also consider updating the labels if the former is not possible.

- The units are now stored internally in metres for almost all units. This should be kept because it is more consistent for the programmer to deal with.
- Any conversion should be done at the final stage before displaying on the view. User-entered values in mm should be immediately converted to m for storage in the model. 
- Some hard-coded values are not stored in metres. Ideally these should also be stored in metres and converted for the view to be consistent with other values, but this may be better done as a maintenance step after release 5.1.
- Note that the beam centre finder results window shows values in mm. These units should probably be the same as the inputs (ideally both mm)

### Suggestions
- The values that have changed from mm to m are shown in the screenshots (please sanity check in case there are more)
- Note that Z offset does not have units in its label. Ideally this should be added.
- The hard-coded values stored in mm are the beam centre radius limits. There may be other hard-coded values also stored in mm.
- It looks like the best place to do the conversions would be in `_set_on_view` and `_set_on_state_model` in `PresenterCommon`. This is used by `RunTabPresenter` and `SettingsAdjustmentPresenter`. Note that `BeamCentrePresenter` has its own version of these functions that perhaps could be consolidated (but I'm not sure if there's a reason it doesn't use the base presenter).

[image]

[image]

[image]

[image]

smk78, 2020-09-11:
> The values that have changed from mm to m are shown in the screenshots (please sanity check in case there are more)

You got the lot!

smk78, 2020-09-11:
Can we also do something about the displayed precision of some values, please.

I would suggest that for anything being displayed as m or mm, 1 d.p. will be adequate _except_ for beam centre coordinates where mm to 3.d.p. would be useful.

closed 2020-09-17 by LamarMoore