mantidproject/mantid #29417: SANS GUI not recognising changed user file

gemmaguest, 2020-09-11:
**Original reporter:** @smk78 

### Expected behavior
Changing the user file should result in changed outputs. If you clear the user file, the reduction should not run.

### Actual behavior
It seems that at least sometimes the outputs for the new user file are not created unless the old outputs are deleted first. If you clear the user file, the reduction still runs - I'm not sure with what inputs!

### Steps to reproduce the behavior
From the discussion on #29358:

All the tests I did were just with a User File specified at the top of the Interface; the User File fields in the table were empty. So I set a different User File in the table and then did Process All. It worked... but the workspaces contained the result I would have expected from the top User File, not the one in the table!

So then I changed the top User File to be the same as that in the table and did Process All. Now I had to get a different result right? Wrong! I got the same result! To be clear, the initial (top) User File specified merged-detectors with radius and Q-limits to suit. The User File in the table was main-detector only with different radius and Q limits. And when I checked settings were for the second User File and not the first. It's like the program has a memory of what it was doing and hasn't realised things have changed. I also note that the Merge Fit/Shift values that appeared in the Options box in the table from the first User File are still there, even though they are not invoked by the second User File.

And, as I just demonstrated to Gemma, if you clear all user files (so none specified at all) you can still happily reduce data!

I'm wondering if this is in some way tied up with the tri-state logic problem?

gemmaguest, 2020-09-14:
@DannyHindson David thinks he's probably the best person to look at this - but let us know if you'd made a start / got anywhere

DannyHindson, 2020-09-14:
No hadn't started. So no problem to reassign to David

DavidFair, 2020-09-15:
@smk78 I read three test scenarios from your description which I tested:

- Clearing the user file at the top of the screen does not reset the GUI settings, so you can reduce with a blank user file which uses the previous valid ones settings

This is errr...."known". It's never been previously reported to be honest, but has likely from when it was first written. I don't think anyone has noticed as they have loaded a valid file afterwards.
So whilst it is a little bit weird, it's not new and I would argue that it's minor enough that we can probably ignore? What's your opinion on this though.

- Loading a global user file, then another

This works as expected - I tested by creating a copy of a user file, changing wavelength limits, then checking settings / running reduction

- A global file and a custom one in the table

Works as expected, I double checked the source code and this was changed recently. The logic was simplified to now: load the user file whilst ignoring any settings in the settings tab. This may cause some subtle changes you may be detailing but I'm not sure? 
Beyond that though it still respects your decision to select a different user file in the table row, albeit more strictly to the definition "the table row uses the custom user file" rather than the grey area we previously had.

However I may have misunderstood what your seeing. Could you walk me through a reproducer if your still seeing this?

smk78, 2020-09-15:
There seems to be an issue with the third case in your reply if the workspace name exists in the ADS. I can reproduce at will so I should show you.

smk78, 2020-09-16:
> There seems to be an issue with the third case in your reply if the workspace name exists in the ADS. I can reproduce at will so I should show you.

After discussion with @DavidFair it seems this is a can of worms and certainly cannot be fixed easily. It is also possibly a bit of an edge case.

The gist is that if you specify a User File in the table and and Process that row, the reduction proceeds and generates output, and overwrites any existing output with the same name. In most cases this is established and expected behaviour.

_But_, in the case of workspaces (as opposed to files) the existing output is overwritten _even if it is in a workspace folder_ and the new reduction should not be creating output in any folders.

So if you, say, do a merged reduction, which generates a merged workspace and the individual LAB/HAB workspaces in a workspace subfolder, _and then_ go and do just a LAB or HAB reduction, it does the new reduction but it appears as if it hasn't, because it creates the new LAB/HAB workspace in the existing subfolder and not in the top level as it would if it had performed the reduction afresh. This can be proved by either renaming or deleting the workspace that will get overwritten _before_ doing the new reduction.

closed 2021-03-23 by stale[bot]