mantidproject/mantid #29423: Why no "Clear all memory" in Workbench?

smk78, 2020-09-11:
@smk78 asks on behalf of a tester: Is there a reason why the "File" > "Clear all memory" option in MantidPlot, which removed all workspaces and plots in one fell swoop, has not been ported to Workbench?

martyngigg, 2020-09-11:
Thanks @smk78. This was an oversight on our part as we'd been focusing so much on the big-ticket items. We'll take a look.

martyngigg, 2020-09-11:
This feature should simply do:
* show a dialog question box and ask a user if they want to clear all workspaces, instruments etc
  * if yes then arrange a call to `FrameworkManager.Instance().clear()`
  * if no then do nothing.

Please add a test in [mainwindowtest.py](https://github.com/mantidproject/mantid/blob/ccb734b1e026dfae9e4e30ba29ecbb2cb0901a1b/qt/applications/workbench/workbench/test/mainwindowtest.py)

gvardany, 2020-09-15:
This was a requirement also for us. Could you also add a button in the workspace widget (next to Delete), called Clear that will trigger the same action?

Matt-Cumber, 2020-09-15:
> This was a requirement also for us. Could you also add a button in the workspace widget (next to Delete), called Clear that will trigger the same action?

@martyngigg I'm happy to also add this if that is okay with you. I shall create a separate issue though as the original has been approved and merged

martyngigg, 2020-09-15:
As long as we can do that without disturbing too much code. The File menu item was easy to adding without disturbing other things. The workspace widget is more involved. I'm happy for a new PR to be created but we need to make sure it is not too risky.

@gvardany Did you have an issue open already?

martyngigg, 2020-09-15:
Another question. `Clear` on the workspaces widget feels like it should only delete workspaces as the button is attached to that widget.

smk78, 2020-09-15:
> Another question/comment. `Clear` on the workspaces widget feels like it should only delete workspaces as the button is attached to that widget?

But you're getting a dialog box telling you what's about to be cleared. Personally I wouldn't be surprised to see plots of data that's being deleted to vanish too (it's what we do in SasView); I would expect that.

gvardany, 2020-09-15:
No issue, apparently it was only in my notes. It is true that the button should clear only the ADS via mtd.clear(), and perhaps not the instrument data service. I don't think there is any risk as long as there is a dialog to confirm.

martyngigg, 2020-09-15:
@smk78 Yes quite right, people get the option. I was meaning that as @gvardany said we clear more than the workspaces/plots with `Clear All` as we cache things like instrument definitions. If we're happy it clears workspaces (the dependent windows will be deleted automatically anyway just like when you delete a single one) then I think that's fine.

Maybe we should then clarify the message for clear all ...

martyngigg, 2020-09-15:
I've created #29457 to look at this. We can work on it but if it looks to risky and will disturb too much code then it'll have to move to next release with workaround being select all with Ctrl+A and pressing Delete.

martyngigg, 2020-09-15:
Closing this as the implementation here is complete.

PR #29438 (2020-09-15): Added Clear All Memory Action To File Menu On Workbench
**Description of work.**
Added a new action in the main File menu called Clear All Memory which makes a call to the frameworkmanager to clear all memory associated with the AlgorithmManager and the Analysis & Instrument data services.

**To test:**

1. Start Mantid Workbench and go to File > Clear All Memory
2. Check pressing Okay here doesn't break anything and you are returned to workbench
3. Create some different workspaces and create a range of plots 
4. Go to File > Clear All Memory and press Cancel and check you are returned to Work [... cut]
closed 2020-09-15 by martyngigg