mantidproject/mantid #29428: Muon Analysis Error when loading a run after using FDA

robertapplin, 2020-09-11:
**Original reporter:** James Lord

### Expected behavior
Data should be loaded without an error message appearing in the logger.

### Actual behavior
The following error message appears in the logger when loading data:
[image]

### Steps to reproduce the behavior
1. Set default instrument to `HIFI`
2. `Interfaces`->`Muon`->`Frequency Domain Analysis`
3. In the runs box, type `169778` and hit enter. Wait for data to load.
4. Keep FDA open. Go to `Interfaces`->`Muon`->`Muon Analysis`
5. In the runs box, type `169824` and hit enter.
6. The error message above appears in the logger. However, the data appears to load anyway.

### Platforms affected
Found on windows nightly build.

AnthonyLim23, 2020-09-14:
This is the same bug as #28865

Harrietbrown, 2020-09-22:
This issue looks like it would be more work to fix than would be worth during the release sprint, and the error message does not interfere with the running of the gui so it can be left for now.

PR #30785 (2021-03-01): Work Around For Bug When Loading A Run In Muon Analysis And Frequency Domain Analysis 
**Description of work.**
An error would be printed when loading a run both in MA and FDA, I've added a check where the error occurred as a work around.

**To test:**
1. Set default instrument to HIFI
2. Interfaces->Muon->Frequency Domain Analysis
3. In the runs box, type 169778 and hit enter. Wait for data to load.
4. Keep FDA open. Go to Interfaces->Muon->Muon Analysis
5. In the runs box, type 169824 and hit enter.
6. There should be no error message now

Fixes #29428

---

#### Reviewer ####

Please commen [... cut]
closed 2021-03-01 by martyngigg