mantidproject/mantid #29430: Muon Analysis - Ties not applied to all spectra when switching to Simultaneous fitting

robertapplin, 2020-09-11:
**Original reporter:** James Lord

### Expected behavior
When switching to simultaneous fitting on Muon Analysis, any current ties (or constraints)  which are present in the selected spectrum should be 'copied' across to all spectra which are to be fitted simultaneously.

### Actual behavior
The other spectra do not have the necessary ties or constraints which have already been applied to the selected spectrum when simultaneous fit is ticked.

### Steps to reproduce the behavior
1. Set default instrument to `HIFI`
2. `Interfaces`->`Muon`->`Muon Analysis`
3. Load runs `169810-169818`
4. Go to `Fitting` tab
5. Tick `Simultaneous fit over`, and then select `Group/Pair` in the drop-down box
6. Untick `Simultaneous fit over`
7. Right click and add function `ExpDecayOsc` to the `FunctionBrowser`
8. Set `Lambda` to zero and `Fix` it
9. Tick `Simultaneous fit over`
10. Switch between spectra using the `>>` button above the `FunctionBrowser`.

Notice that the other spectra do not have Lambda tied to zero.

### Platforms affected
Found on windows nightly build

AnthonyLim23, 2020-09-14:
This is not that surprising, the two browsers are different and we are just switching between the two. I'm not convinced there is much we can do about it.

robertapplin, 2020-09-14:
I think I agree. There is not much we can do unless we try change something in the `FunctionModel`, which I want to avoid doing. This is too minor an issue for such a change.

closed 2020-09-14 by robertapplin