mantidproject/mantid #29433: Investigate 'A plot was unable to be saved' warning

robertapplin, 2020-09-14:
**Original reporter:** James Lord

### Description
A warning message appears in the logger `A plot was unable to be saved` after you plot a scatter plot with y errors. This may be working as intended, but needs investigating or insight from someone who knows more about project save.

### Steps to reproduce the behavior
1. Set default instrument to `HIFI`
1. `Interfaces`->`Muon`->`Muon Analysis`
2. Load `169779-169790`
3. Go to the `Fitting` tab. Right click and add function `ExpDecayMuon`
4. Go to the `Sequential Fitting` tab. Click `Sequentially fit all`
5. Go to `Results` tab and click `Output Results`
6. Look for the `ResultsTable` workspace in the ADS and open it
7. Right click `workspace_name` column and `Set as X`
8. Right click `AError` column and `Set error for Y0`
9. Right click `A[Y0]` column and `Plot scatter with **y errors**`
10. Wait for a minute and the following message appears in the message log:
[image]

### Platforms affected
Tested on windows nightly build.

Pasarus, 2020-09-15:
- Load this table workspace:
[scatterabletable.zip](https://github.com/mantidproject/mantid/files/5223875/scatterabletable.zip)
- Open data view
- Right click workspace_name column and Set as X
- Right click AError column and Set error for Y0
- Right click A[Y0] column and Plot scatter with y errors
- Save project or Wait for project recovery to flop over this plot

Pasarus, 2020-09-15:
This issue has been made into 3 and assigned to the relevant teams:
- https://github.com/mantidproject/mantid/issues/29463
- https://github.com/mantidproject/mantid/issues/29462
- https://github.com/mantidproject/mantid/issues/29461

closed 2020-09-15 by Pasarus