mantidproject/mantid #29455: Issues with Training Courses

DanielMurphy22, 2020-09-15:
Found by @sf1919 

- [ ] [getName method has been replaced with name](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/algorithms_and_workspaces/02_ws_types.html)

sf1919, 2020-09-18:
In the Overall Example Script on https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/further_alg_ws/02_scripting_workspaces.html there is a line commented out - `#plotSpectrum(data,[1,2,3],error_bars=True, waterfall=False)` . Does this need to be there? There's nothing in the Output that suggests it needs to be there.

sf1919, 2020-09-23:
In Declaring Array Propertie (https://www.mantidproject.org/Declaring_Array_Properties) all the links in the Validation section (e.g. FloatArrayLengthValidator) return 404 errors.

DanielMurphy22, 2020-09-29:
https://www.mantidproject.org/Python_Exercise_Two_Solution

PR #29716 (2020-10-15): Mantid Basic Course Updates
**Description of work.**
Updated images related to Plots and Settings
Improved instructions and explanations in specific parts of the Basic Course 
Also aided understanding with a change to ConvertUnits docs

**To test:**

Build docs-html
Check that the modified pages look sensible and the new explanations / images are appropriate!
See #29455 for more details.

Fixes #29455 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an accep [... cut]
closed 2020-10-15 by martyngigg