mantidproject/mantid #29459: ALCResults Workspace structure and Plots do not survive project save

DanielMurphy22, 2020-09-15:
**Original reporter:** James Lord

### Expected behavior
Save and reloading a Project with a workspace and plot should work
### Actual behavior
Exported workspaces (and plots) from ALC interface don't survive project save/load
### Steps to reproduce the behavior
- Open ALC (with ISIS archive access)
- Type in some valid run number(s) [e.g. 50000] and select `Load` at the bottom 
- Once loaded, a scatter plot will appear 
- Click `Export Results` below the plot to save out to a workspace.
- By default it saves to a WorkspaceGroup called `ALCResults`, containing one workspace: `ALCResults_Loaded_Data`
[_This Workspace group structure seems odd and the right-click menus for the group and the contained workspace are quite different_]
- Right-click and `Plot>Spectrum` either the workspace group or the contained workspace [both actually work on the contained ws].
- Finally save this as a project
- Delete all workspaces
- Open the project.
- The workspace group has been replaced with a normal workspace (the contained one, just not in a group, but it now has the same name as the group: `ALCResults`) and Project save is unable to reproduce the plot, as : `Error: "'ALCResults_Loaded_Data' does not exist."`
### Platforms affected
Found in Beta testing 5.0.20200909.1539, Windows 10 BUT also exists in release 5.0

Matt-Cumber, 2020-11-13:
@DanielMurphy22 This isn't just from the ALC. This is generally happening in workbench if you have a group workspace with only one workspace within it. It always get's renamed and the group is removed on project load

DanielMurphy22, 2020-11-13:
@Matt-Cumber I agree that this is the case and I believe this is the intended behaviour of group workspaces with only one contained workspace. I guess its annoying that the ALC interface produces this group with one workspace in the first place. 

I understand why both the Core and Spec teams think this is work for the other team to do! Either project load or the output from ALC should be changed, but maybe this is a discussion for the team leads.

sf1919, 2021-11-25:
@AnthonyLim23 is this one that should have the Roadmap label?

closed 2022-06-10 by stale[bot]