mantidproject/mantid #29464: IndirectDataAnalysis should have a warning if the spectra number of sample is different to the resolution

Harrietbrown, 2020-09-15:
In indirect Data analysis I(Q,t) and Convfit there should be a warning if the sample and resolution workspace have different number of spectra.

SpencerHowells, 2020-09-15:
@Harrietbrown The resolution can be a red or res file. The res file has only 1 spectrum & the program makes a copy of this spectrum for each spectrum of the sample. For a resolution red file the prog checks that the  number of spectra in sample and resolution are the same - it certainly used to!

SpencerHowells, 2020-09-15:
@Harrietbrown Iqt runs alg TransformToIqt. This has
        # Process resolution data
        res_number_of_histograms = CheckHistZero(self._resolution)[0]
        sample_number_of_histograms = CheckHistZero(self._sample)[0]
        if res_number_of_histograms > 1 and sample_number_of_histograms is not res_number_of_histograms:
            CheckHistSame( self._sample,  'Sample',  self._resolution,  'Resolution')
Then CheckHistSame has
    num_hist_1 = s_api.mtd[in1WS].getNumberHistograms()  # no. of hist/groups in WS1
    x_1 = s_api.mtd[in1WS].readX(0)
    x_len_1 = len(x_1)
    num_hist_2 = s_api.mtd[in2WS].getNumberHistograms()  # no. of hist/groups in WS2
    x_2 = s_api.mtd[in2WS].readX(0)
    x_len_2 = len(x_2)
    if num_hist_1 != num_hist_2:  # Check that no. groups are the same
        error_1 = '%s (%s) histograms (%d)' % (name1, in1WS, num_hist_1)
        error_2 = '%s (%s) histograms (%d)' % (name2, in2WS, num_hist_2)
        error = error_1 + ' not = ' + error_2
        raise ValueError(error)
    elif x_len_1 != x_len_2:
        error_1 = '%s (%s) array length (%d)' % (name1, in1WS, x_len_1)
        error_2 = '%s (%s) array length (%d)' % (name2, in2WS, x_len_2)
        error = error_1 + ' not = ' + error_2
        raise ValueError(error)

AnthonyLim23, 2021-04-28:
@Harrietbrown does this need to be fixed for this release?

SpencerHowells, 2021-04-28:
@AnthonyLim23 @Harrietbrown See my comment above. It already does the appropraite checks.