mantidproject/mantid #29465: Instrument View shrink fix on MacOS

DanielMurphy22, 2020-09-15:
Leads on from #27244 which was fixed by #29432

This issue is to investigate why building a package for MacOS solves the #27244 issue. 

- A package after (_and even before_) this PR seems to have fixed this issue on MacOS. 
- Oddly a local build (not converted into a package) fails in all cases.
- Launching a package from terminal is still fixed.

It is not very satisfying that the build is broken and the package is fixed. 
While testing if the window shrink issue occurs is the ultimate test, running this script is also useful:
```
from qtpy.QtWidgets import QApplication
for screen in QApplication.screens():
    print(f"devicePixelRatio: {screen.devicePixelRatio()}")
    print(f"logicalDotsPerInch: {screen.logicalDotsPerInch()}")
    print(f"physicalDotsPerInch: {screen.physicalDotsPerInch()}")
```
On the package (fine) I get:
```
devicePixelRatio: 
1.0
logicalDotsPerInch: 
72.0
physicalDotsPerInch: 
129.00000193738563
devicePixelRatio: 
1.0
logicalDotsPerInch: 
72.0
physicalDotsPerInch: 
94.00000141173837
```
And on the build (broken) I get:
```
devicePixelRatio: 
2.0
logicalDotsPerInch: 
72.0
physicalDotsPerInch: 
129.00000193738563
devicePixelRatio: 
1.0
logicalDotsPerInch: 
72.0
physicalDotsPerInch: 
94.00000141173837
```

[A pixel ratio of 2.0 is what led to the quarter window shrink in #27244, which is handled on windows by the PR #29432. On Mac, it being fixed seems magically unrelated to these code cahnges]

DanielMurphy22, 2020-09-15:
Could this be linked to #28468

martyngigg, 2020-09-29:
@DanielMurphy22 I think we have fixed this now? Or did we leave this open because your developer build still misbehaves?

DanielMurphy22, 2020-09-29:
Yes I created this to solve the mystery as to why it "magically" works in a package but not a build. But this is not release blocking. I'll assign myself to remove confusion!

martyngigg, 2020-10-27:
Closing as macOS handes the scaling at the OS level according to the Qt docs.

closed 2020-10-27 by martyngigg