mantidproject/mantid #29469: Attempting to Load file while another is already loading causes a crash

robertapplin, 2020-09-15:
**Original reporter:** David Voneshen

### Expected behavior
The second file for loading should be queued for loading, and should start loading when the previous file has finished loading.

### Actual behavior
Mantid crashes.

### Steps to reproduce the behavior
1. Above the ADS use `Load` to load a file
2. Quickly click `Load` and load another file before the previous file has finished loading.

**Data File**
Any large file will work (I think), but you can try using this one:
[testData.zip](https://github.com/mantidproject/mantid/files/5225991/testData.zip)

### Platforms affected
Testing on Windows and linux nightly builds

martyngigg, 2020-09-16:
@robertapplin Do we know if this is only NeXus files?

robertapplin, 2020-09-16:
> @robertapplin Do we know if this is only NeXus files?
From what I have seen it is only NeXus files. Raw seem to be fine.

martyngigg, 2020-09-18:
I was able to extract a stacktrace with [WinDbg](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools):
```
[0x0]   hdf5!H5C_protect + 0x811   
[0x1]   hdf5!H5AC_protect + 0xc3   
[0x2]   hdf5!H5O_protect + 0x179   
[0x3]   hdf5!H5O_msg_exists + 0xdc   
[0x4]   hdf5!H5D__layout_oh_read + 0x3f   
[0x5]   hdf5!H5D__open_name + 0x3e6   
[0x6]   hdf5!H5D_open + 0x223   
[0x7]   hdf5!H5D__open_name + 0x114   
[0x8]   hdf5!H5Dopen2 + 0x171   
[0x9]   libNeXus_0!nxifputattr_ + 0x6510   
[0xa]   libNeXus_0!nxiopendata_ + 0x8b   
[0xb]   libNeXusCPP_0!NeXus::File::openData + 0x9b   
[0xc]   MantidKernel!Mantid::Kernel::PropertyNexus::loadProperty + 0x60d   
[0xd]   MantidAPI!Mantid::API::LogManager::loadNexus + 0xb3   
[0xe]   MantidAPI!Mantid::API::Run::loadNexus + 0x11f   
[0xf]   MantidAPI!Mantid::API::ExperimentInfo::loadSampleAndLogInfoNexus + 0x27b   
[0x10]   MantidAPI!Mantid::API::ExperimentInfo::loadExperimentInfoNexus + 0x28   
[0x11]   MantidDataHandling!Mantid::DataHandling::LoadNexusProcessed::loadEntry + 0x1893   
[0x12]   MantidDataHandling!Mantid::DataHandling::LoadNexusProcessed::exec + 0x1139   
[0x13]   MantidAPI!Mantid::API::Algorithm::exec + 0x80   
[0x14]   MantidAPI!Mantid::API::Algorithm::executeInternal + 0x7b6   
[0x15]   MantidDataHandling!Mantid::DataHandling::Load::loadSingleFile + 0x6b8   
[0x16]   MantidDataHandling!Mantid::DataHandling::Load::exec + 0x21f   
[0x17]   MantidAPI!Mantid::API::Algorithm::exec + 0x80   
[0x18]   MantidAPI!Mantid::API::Algorithm::executeInternal + 0x7b6   
[0x19]   MantidAPI!Mantid::API::Algorithm::executeAsyncImpl + 0x23   
[0x1a]   MantidAPI!Mantid::Kernel::DataService<Mantid::Geometry::Instrument>::retrieve + 0x195   
[0x1b]   PocoFoundation64!Poco::ThreadPool::joinAll + 0x30b   
[0x1c]   PocoFoundation64!Poco::ThreadImpl::runnableEntry + 0x51   
[0x1d]   KERNEL32!BaseThreadInitThunk + 0x14   
[0x1e]   ntdll!RtlUserThreadStart + 0x21   
```
This looks very much like some issue with the multithreaded access to the HDF libraries.

martyngigg, 2020-09-18:
Marked for 5.1 has it's a regression from 5.0.

PR #29542 (2020-09-22): Fix loading NeXus file while second is loading
**Description of work.**

Pulls in a rebuild of the HDF5 libraries on Windows that have the threadsafe option enabled. Without this option it is not possible to access the HDF5 libraries safely from multiple threads.

**To test:**

* Build on Windows in release mode
* Attempt the steps in the related issue and on clicking Load for the second time mantid should not crash.

Fixes #29469  

*This does not require release notes* because **it is a regression from v5.0**

---

#### Reviewer ####

Please comment on the following ([full description](http: [... cut]
closed 2020-09-22 by martyngigg