mantidproject/mantid #29498: Easy simulated muon data

AnthonyLim23, 2020-09-16:
Want to make it easy for users to produce simulated muon data (including log files). We can just copy log files etc. from a different run

James should be able to help.

AnthonyLim23, 2021-07-20:
We will need to create a new algorithm with the following options:
Option to copy sample logs from a real run, otherwise to use default values
To change the dead times, alpha, time zero’s etc.
To handle a pseudo instrument
Most of the simulated data can be produced from a script written by James. It is just missing the above functionality. 

The simulated data is defined by a fitting function, which may come from Quantum or somewhere else, and is then modified to reflect the raw data. This allows users to see what set up they may need for a given experiment and for how long (by altering the counting stats). They may also simulate new ideas for instruments.

From James:
I have a “Simulated muon data” algorithm already in the Repository, under “muon”: SimulateMuonTFData.py
There are two algorithms in there. The first, “SetupSimInstrument”, defines the instrument and sample setup. It sets alpha, detector count rates, dead times, pulse width to convolute with, etc and puts them in a configuration table. It works from the instrument definitions but could be modified to generate an arbitrary instrument.
The second algorithm “SimulateMuonTFData” takes this configuration table and some details of the measurement being made and generates a simulated data workspace. This version uses time bins as defined in the table, and a mathematical formula for the asymmetry (some predefined ones are available). It could easily be rewritten to take a workspace with A(t) and use the X bins of that as the X for the simulated workspace. Lines 409-426 do the actual generation of the data.

You could imagine doing:
-	Set up a simulated instrument
-	Simulate a high stats ZF run and run “CalMuonDeadTime” on it, and keep these dead times for use in Muon Analysis (they ought to be close to the randomly chosen ones during setup)
-	Simulate a TF 20G run and use it to guess Alpha (again should be close to the setup alpha)
-	Simulate some data with a relaxation function of your choice, and some value of total counts. Fit it in the Muon Analysis interface.

For the “experiment design” use case, I might want to ask:
-	Given this relaxation function, what stats do I need to measure lambda (or other parameter) to an accuracy of say 1%
-	Given two alternative models and their relaxation functions, what stats do I need in order to tell which of them best fits this sample with some certainty?

martyngigg, 2021-08-23:
@AnthonyLim23 This is still assigned to Matt. Could you move it to the relevant milestone?

closed 2023-02-19 by stale[bot]