mantidproject/mantid #29500: Potential Updates To ALC Interface

Matt-Cumber, 2020-09-16:
**Original reporter:** Stephen Cottrell

Creating this issue as a reminder for the next meeting as to what will go into the next release.

### Small Requests

1. Adding an X label to the plot on the data loading interface
2. Automatically setting the default in periods to 1 subtract 2 if there are only 2 periods
3. Making load button unavailable until a valid runs expression is entered and has successfully updated (found sometimes could change runs but still load the same data not new data)

### Bigger Requests

1. Re-add the old watcher functionality to the runs box when auto is ticked so that as new runs are added to the watched directory the data is updated
2. Separate runs into a path box and a runs box so a user can set a path first then choose runs from that path
3. Load time improvements if possible

AnthonyLim23, 2020-10-06:
Add an instrument option

Matt-Cumber, 2020-10-14:
Warning box if runs > 200

Matt-Cumber, 2020-11-06:
colour code status messages

Matt-Cumber, 2020-11-06:
attempting to find files message needs to be clearer

closed 2020-11-30 by SimonHeybrock