mantidproject/mantid #29504: Workbench not saving user preferences

DavidFair, 2020-09-17:
### Expected behavior

Any changes settings should be preserved between opening workbench

### Actual behavior

Settings appear to be lost, causing users to get into the habit of loading in their own properties file each launch

### Steps to reproduce the behavior

*Note: This is part speculation on how it's happened*

Either:
- Open workbench
- In File -> Settings, change all the preferences making note (e.g. through screenshots) of the new options
- Close workbench
- Check all the settings persist and there aren't one or two getting left behind

-----
Alternatively, this is caused by loading a properties file on another machine not being persisted

- Open workbench
- In File -> Settings, change all preferences making note again
- Save to properties file
- Revert settings back to original value (or open on another machine)
- Load in properties file
- Close and open Mantid, the settings should persist to the settings you noted, not the default ones

### Platforms affected
Windows, possibly all

RichardWaiteSTFC, 2020-09-17:
I can't seem to reproduce this on nightly from 14 Sep. I changed options in all settings categories and it remembered after closing and starting again. I think one source of confusion might be that you have to click "close" to update the settings. You can save/load a options to/from a different fiel but mantid will always load the default (i.e. not the one you created). Maybe the user wasn't clicking close?

DavidFair, 2020-09-17:
>  You can save/load a options to/from a different file but mantid will always load the default (i.e. not the one you created).

Sorry I'm a bit confused by this line, if you load in a file does that mean Mantid will load in the default next time?

RichardWaiteSTFC, 2020-09-17:
> Sorry I'm a bit confused by this line, if you load in a file does that mean Mantid will load in the default next time?
Ah sorry my mistake - it's just because I managed to corrupt the properties file badly but can't reproduce it... Now I've got a valid properties file it does seem to automatically load in the file you just saved. So can't reproduce the problem. Still think "Close"
 should be renamed to "update" or "apply"

DavidFair, 2020-09-17:
So if you close the page without using the close button it won't apply?

If so I think renaming "Close" -> "Update" is a risk free fix that makes it clear you need to hit that rather than using the window close button - and avoids us messing with signals and slots

Longer term, we should wire up closing the window to saving the settings

RichardWaiteSTFC, 2020-09-17:
> So if you close the page without using the close button it won't apply?

...good point - it does indeed sorry! So the close button is appropriately labelled but is this the expected behaviour? I was expecting (and I should have checked) that I could cancel any of my changes if I hadn't clicked the buttons...this might be useful if I was unsure about a change (i think this is typically how menus work in mantid?) Perhaps it would be nice to have a button to restore defaults?

DavidFair, 2020-09-17:
So IMO:

- For this release, rename `close` to `save` or similar

For the next release, either
- Rename `close` to `apply` (which keeps the settings page open)
- Or have `Save` and `Cancel` Buttons

DavidFair, 2020-09-17:
So talking to Richard we can't reproduce this currently. I suspect that it might be related to a read-only bit which @Pasarus ran into before.
Qt will "successfully" write out the settings, but not commit the changes to the file so they are lost each time.

Either way this is worth pushing off to a future release

DavidFair, 2020-09-17:
I'm going to close this until we either get more reports, or a reliable reproducer too

closed 2020-09-17 by DavidFair