mantidproject/mantid #29506: Create a dedicated sliceviewer release notes page

martyngigg, 2020-09-17:
As a developer we want to announce developments to users. The slicevewer has had many changes over this release cycle and as a result it deserves its own sub page of the workbench release notes.

It should be image heavy to show off all of the new features including:

* non-orthogonal view including peak overlays
* roi/line-cut selection and extraction
* general UX improvements

DanielMurphy22, 2020-09-17:
There are some additional improvements from others in the team that are currently in mantidworkbench.rst and should be added to this page. Happy to help out where I can for this!

DanielMurphy22, 2020-09-23:
The release notes from MantidWorkbench.rst that should be added to this new page are:

```
- The Sliceviewer interface has been neatened up to reduce wasted space and devote more to the data view itself.

- Sliceviewer now contains an interactive table of plot details that updates as you move your cursor over the data (functionality from the SpectrumViewer in Mantidplot). This table includes the dimension value and signal for multidimensional workspaces, and for matrix workspaces with a spectrum axis:
   - Signal
   - Spectra Number
   - Detector ID
   - Two Theta
   - Azimuthal Angle
   - Time of Flight
   - Wavelength
   - Energy
   - d-Spacing
   - Mod Q

- On opening, the zoom tool is selected by default.
- Reversed colourmaps are now accessed with a reverse checkbox.
- The default colourmap can be set in File > Settings.
- The scale (lin/log) remains set when reopened.

- Sub-plots in the sliceviewer now follow the scaling on the colorbar.
- When line plots are active on the sliceviewer the arrow keys can now be used to move the cursor a pixel at a time.
- The line width of the line plots has been reduced.

- Figure options have been removed as most options were not appropriate.
```

DanielMurphy22, 2020-09-24:
For my own reference:

- SliceViewer can now correctly display non-orthogonal axes for output of the MDNorm algorithm.

DavidFair, 2020-09-25:
Closed by #29591

PR #29591 (2020-09-25): Add separate page for Sliceviewer release notes
**Description of work.**

Adds a separate release notes page for the sliceviewer. It has had a significant number of new features that are worth highlighting on their own.

**To test:**

Build the docs and check they look okay with all of the images included.

Fixes #29506 

*This does not require release notes* because **it is release notes**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable qualit [... cut]
closed 2020-09-25 by DavidFair