mantidproject/mantid #29522: Sliceviewer data table text not contrasting enough

DanielMurphy22, 2020-09-18:
@LamarMoore @martyngigg is this worth putting in this release?? We could just make this text black??

**Original reporter:** Spencer on the Forum: https://forum.mantidproject.org/t/grey-level-in-ubuntu/689

Grey text in the interactive data table in the new workbench sliceviewer is okay on Windows,
[image]

 but much harder to read on MacOS/Linux (Not contrasting enough)

[image]
[image]

SpencerHowells, 2020-09-18:
@DanielMurphy22 @LamarMoore @martyngigg this occurs everywhere. See also the info when you click on the ws
[image]

martyngigg, 2020-09-18:
@SpencerHowells We're not explicitly setting this so it must be the Qt style defaults for the type of item that is displayed. I'd prefer not to make a wholesale change at this point for the application but the table in the sliceviewer is actually doing it more by accident than design and I think we can at least target a small fix for that for this release.

martyngigg, 2020-09-18:
We also need to be careful messing with styles as it's possible other Linu distros set different styles and what we set for a specific colour might clash. We can look into it though.

PR #29557 (2020-09-23): Mark cells in Sliceviewer ImageInfoWidget read only and remove selectable flag
**Description of work.**

The text colour in the cursor info widget was quite hard to read and had accidentally been set this way. Removing the selectable flag means the text is now black rather than very light gray. This change also removes the editable flag as it was not sensible to edit the table.

**To test:**

* Load any MatrixWorkspace or MDWorkspace from the test data
* Open the slice viewer
* The cursor info table should now have black text and no longer be editable 

Fixes #29522

*This does not require rel [... cut]
closed 2020-09-24 by LamarMoore