mantidproject/mantid #29554: Sometimes crash from Indirect reduction

AnthonyLim23, 2020-09-21:
For indirect they have 2 methods of storing their data; nxs and raw. The raw files are accompanied by the following files status, ICPrevent, ICPStatus and ICPdebug. 

It is possible to crash workbench when these additional files are not present. However, it is not 100% reproducible (there is something else going on). I can reproduce it on my laptop but not my desktop

Turn off the data archive
Open indirect data reduction
Load the raw file (with the other files missing) using browse of the run bar
If it is going to crash it will here

AnthonyLim23, 2020-09-21:
This issue was in version 5

SpencerHowells, 2020-09-21:
@AnthonyLim23 the clue is in the phrase ‘with the other files missing’! The user must put them there.

PR #29574 (2020-09-24): Avoid hard crash in Indirect Energy Transfer load without TCP files
**Description of work.**
Add a check to avoid Energy Transfer from trying to access files from the ADS that haven't loaded correctly, letting it fail 'nicely' like the other tabs rather than hard crashing

**Report to:** Sanghamitra, Anthony

**To test:**
1. Turn off the data archive, download an IRIS .raw but NOT the three accompanying ICP files (e.g. IRIS00055127)
2. Interfaces -> Indirect -> Data reduction -> Energy transfer (default tab)
3. Input runs -> browse to the .raw and try to load it

Fixes #29554. 

---

#### Revi [... cut]
closed 2020-09-24 by martyngigg