mantidproject/mantid #29602: Clean/Dirty Install Smoke Tests

DanielMurphy22, 2020-09-25:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files
* Windows:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Clean out the registry
  * Load regedit (Command Prompt > regedit)
  * Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it
* OSX:
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`
* Linux
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

DanielMurphy22, 2020-09-25:
@darominski Error Report: Mantid workbench crashed when trying to invoke 'Show instrument' on an empty instrument, in this case D7.

gui-co, 2020-09-25:
On MacOS, if two MantidPlot are installed, I am only able to start the one that has been installed first.

DanielMurphy22, 2020-09-25:
> On MacOS, if two MantidPlot are installed, I am only able to start the one that has been installed first.

When installing, instead of replacing are you `Keeping Both`??

darominski, 2020-09-25:
On Windows: No apparent version clash for the dirty install. Mantid Workbench and MantidPlot start fine. Clear install is also fine.

However, in both clean and dirty install, the Workbench  crashes when I start the instrument viewer, tested for D7, D16, ISIS example data (MUSR00015189). This does not happen for the MantidPlot. Also, I did not observe any other issues with plotting in the Workbench, for neither 1D bin plot,  spectrum plot, nor the colorfill.

gui-co, 2020-09-25:
> When installing, instead of replacing are you `Keeping Both`??
Yes, keeping both.

MathieuTillet, 2020-09-25:
No issue encountered on Ubuntu 18.04.

DanielMurphy22, 2020-09-25:
@darominski 
> However, in both clean and dirty install, the Workbench crashes when I start the instrument viewer, tested for D7, D16, ISIS example data (MUSR00015189). This does not happen for the MantidPlot. [...]

I have tried to reproduce this on Windows (with MUSR00015189.nxs and MUSR00015189.nxs) on Windows. Are you able to point me towards relevant D7 or D16 data files??

closed 2020-09-30 by martyngigg