mantidproject/mantid #29603: Fitting Smoke Tests

DanielMurphy22, 2020-09-25:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_fit_model_choices.html#fit-model-choices)
- [ ] Non peak model + background 
- [ ] User defined function

darominski, 2020-09-25:
Windows: No issues found at any step.

DanielMurphy22, 2020-09-25:
Found problems on Mac:

- Trying to add other function and then typing to search for a function produces unhandled exception: 
```
 File "/private/var/folders/br/23hyhlhn1l3c5fqs0v_t9r_40000gn/T/AppTranslocation/E9975322-647D-445D-AB11-0A592EE991A0/d/MantidWorkbench.app/Contents/MacOS/mantidqt/widgets/fitpropertybrowser/addfunctiondialog/view.py", line 59, in keyPressEvent
   self.lineEdit().keyPressEvent(event)
RuntimeError: no access to protected functions or signals for objects not created from Python
```

- Tried fitting with a UserFunction of r*exp(a) which did not work : 
```
Error in execution of algorithm Fit:
Error evaluating function: Unexpected end of expression at position 0
```

closed 2020-09-30 by martyngigg