mantidproject/mantid #29604: Interface Smoke Tests

DanielMurphy22, 2020-09-25:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

- [ ] Mantid opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Catalogue opens  [directions here](https://www.mantidproject.org/Catalog_Search) 
- [ ] Saving/loading projects works  [directions here](https://www.mantidproject.org/MantidPlot:_The_File_Menu#File-.3E_Save_Project_as...)

sf1919, 2020-09-25:
When opening Mantid Workbench with facility as ILL and then trying to open the ILL->Drill Interface an error message is encountered. The error message is `Instrument IN4 is not supported yet.` However an interface window does open as per screen shot below. Is this expected behaviour?

[image]

sf1919, 2020-09-25:
The interface Utility ->TofConverter opens up very small and unlike other interfaces will not allow me to resize it. It's very difficult to read/use.
[image]

DavidFair, 2020-09-25:
Testing on Mantid 18.04, worked through everything above.

Things noticed:

- Switching the facility to HFIR then ILL loses the ILL Drill Interface from the drop down for good

- The warnings @sf1919 saw about unsupported instruments come from the Drill interface when you switch facility, so I think this is obscure enough not to worry about (as they well, are unsupported in that interface)
- There was some minor crushing of text boxes in some Mantid Plot interfaces, but it was only 2-3px and still very readable so again I think we can ignore it

sf1919, 2020-09-25:
Testing on Mantid for Windows with a dual 4K/2K monitor setup, worked through everything above on each monitor.

Things noticed:

- In Workbench there is a warning regarding the Drill Interface (outlined above)
- It is not possible to resize the TofConverter Interface in Workbench (outlined above). The Interface is a better size and therefore easier to read/use in MantidPlot
- In MantidPlot when the Old ISIS SANS Interface is opened there is an immediate user file extension error message - before any data is loaded.
- Having tested on both screens issues #29596 and #27244 which I had previously encountered were no longer happening

gvardany, 2020-09-25:
If the default instrument is not among the supported ones there will be a message, but perhaps error is too high. Then the instrument selector falls back to one of the supported ones. We are looking into this. However could you please test the following:
if you set the instrument (in the mantid global settings) to D11, it should open up with no error.
> In Workbench there is a warning regarding the Drill Interface (outlined above)

sf1919, 2020-09-25:
I can confirm if you set the Instrument to D11 in the global settings then the Interface does open without an error message - at least on Windows 10.

DanielMurphy22, 2020-09-25:
Works fine on MacOS apart from Elemental Analysis has crushing issues on opening, but this bug has been around for a few releases.

gui-co, 2020-09-25:
I made some changes related to what @sf1919 mentioned -> #29618 
Interface behaviour should be clearer now when an unsupported instrument is selected.

gvardany, 2020-09-25:
The approved PR is here, can anyone merge it? https://github.com/mantidproject/mantid/pull/29618#pullrequestreview-496640629

closed 2020-09-30 by martyngigg