mantidproject/mantid #29605: Plotting Smoke Tests

DanielMurphy22, 2020-09-25:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Plot from matrix 
 - [ ] Change colour map 
 - [ ] Contour plot 
 - [ ] Spectrum viewer 
* 3D plotting:
 - Load some data eg `LOQ7041` from the ISIS sample data
 - Double click the workspace to open the table view
 - Select the table
 - [ ] 3D plot becomes available as an option in the toolbar 
 - [ ] 3D wire frame 
 - [ ] 3D hidden line 
 - [ ] 3D polygon 
 - [ ] 3D wire surface

MathieuTillet, 2020-09-25:
Using both Workbench and Mantidplot on Ubuntu 18.04, my plots for the 1D plotting looks not at all like the ones provided in the instructions, but look consistent with the data itself.
(Also, Workbench's loader does not show `.raw` data, which is inconvenient for these examples)

Harrietbrown, 2020-09-25:
All the plotting systems work as we would want. the autogenorated issue for this should be updated on how to create a 3D plot since you can get to them by the plotting menu now.

MathieuTillet, 2020-09-25:
On Ubuntu 18.04, everything works as intended, except for 2 small issues on Workbench :
- when opening the figure options of the colorfill plot, a few warnings about the Vega20b color map deprecated are emitted.
- when opening the slice viewer with Autoscaling checked, right-clicking the image once change the scale range, and right-clicking it again change it again to yet another value, and then right-clicking more does not change anything anymore.

closed 2020-09-30 by martyngigg