mantidproject/mantid #29606: Python Smoke Tests

DanielMurphy22, 2020-09-25:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* Test that the Python scripting window works, [directions here](http://docs.mantidproject.org/v3.7.2/interfaces/ScriptingWindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](http://www.mantidproject.org/Python_In_Mantid), 3 or 4 examples is enough

clayton-tom, 2020-09-25:
Plot: Using Comment in the script editor removes the first character of the line underneath the line being commented

DavidFair, 2020-09-25:
Ubuntu 18.04:

(@DanielMurphy22 Some of the solutions in https://www.mantidproject.org/SolutionsToPythonInMantid need updating for workbench)

Workbench:
- Run all only runs the current script window, not all (both were unsaved) - #29616 

Mantid Plot:
- I'm not seeing the issue Tom is with comment / uncomment, so it is possibly Windows(?) only

gui-co, 2020-09-25:
### On macOS ###

#### Workbench ####
* when opening a second script tab (an a third, etc...), the font is much bigger in that tab

#### MantidPlot ####
* my keymap is mixed up, especially the top row (numbers). I worked through VNC, it could be linked to that. The problem is only present in MantidPlot. I checked the last release and it is also the case
* the problem that Tom mentioned is not present on macOS

closed 2020-09-30 by martyngigg