mantidproject/mantid #29616: Workbench Run All only runs current window

DavidFair, 2020-09-25:
I've tagged this for 5.1, I'm on the fence if it's important enough to fix though.
On the one hand the probability of a user running into it is Medium. But the impact is low.

I'll let someone more familiar with the code triage this between 6.0 and 5.1

### Expected behavior

Run all should run all windows

### Actual behavior

Only the current window is run

### Steps to reproduce the behavior

- Open two script windows, don't save (might be relevant)
- Ensure the output window is set to output from both
- Enter the following with `running` set to a different value

```python
import time

while(True):
    print("running2")
    time.sleep(0.5)
```
- Hit run all, only the current window runs

### Platforms affected
Ubuntu, possibly all?

clayton-tom, 2020-09-25:
I may have misunderstood, but I'd thought `Run` should execute the current selection, or whole script if no selection is made, whereas `Run All` runs the entire script. Seems to be a change in wording but not in functionality from Plot https://docs.mantidproject.org/v3.7.2/interfaces/ScriptingWindow.html

DavidFair, 2020-09-25:
Ahh right so it's Mantid Plot's execute selection / execute all

As it's next to the run button for the current script I assumed it ran all scripts. In this case lets close it and if users get confused we can come back to it.

closed 2020-09-25 by DavidFair