mantidproject/mantid #29623: Fitting with BivariateGoussian Crashes Workbench

Harrietbrown, 2020-09-28:
### Expected behavior
After attempting to fit with BivariateGoussian, even if the fit fails you should be able to fit with a different function
### Actual behavior
When fitting with BivariateGoussian an error is raised and if you attempt to fit again mantid is crashed
### Steps to reproduce the behavior
1. Load MAR11015
2. plot a single spectra
3. open the fitting browser
4. add BivariateGaussian function
5. Try to fit an it will fail with an error message
6. Remove function and add a flat background.
7. click fit agian and mantid will Crash
### Platforms affected
Found in windows.

DanielMurphy22, 2020-09-28:
This is a problem in the nightly, and release 5.0. I'd just like to add that when I ran BivariateGaussian with a LinearBackground I got this output: 
```
Error in execution of algorithm Fit:
cannot reshape array of size 2656 into shape (50,50,2)
  at line 98 in 'C:/MantidInstall/plugins/python\functions/BivariateGaussian.py'
```

Harrietbrown, 2020-09-28:
That was the error message I reffered to in step 5

PR #29647 (2020-10-01): Update BivariateGaussian documentation and throw warning for improper inputs
**Description of work.**
This PR updates the BivariateGaussian documentation as it referenced a removed matplotlib function. The function also now throws a warning if the input array cannot be reshaped into the desired (nX * nY * 2) matrix.

**To test:**
1. Check that the example in the documentation works
2. To replicate the problem in the issue load some data e.g MUSR00062260.nxs
3. Plot a spectrum
4. Open the fit browser
5. Add a BivariateGaussian function
6. Click fit - it should throw the new warning about settin [... cut]
closed 2020-10-01 by martyngigg