mantidproject/mantid #29636: Show Detectors hang on TOPAZ data

DanielMurphy22, 2020-09-29:
Raised in smoke testing by @DannyHindson : https://github.com/mantidproject/mantid/issues/29607#issuecomment-699878909

### Expected behavior
Show detectors for all instruments works quickly.
### Actual behavior
Showing detectors on TOPAZ_3132_event.nxs hangs
### Steps to reproduce the behavior
- Load TOPAZ_3132_event.nxs and right-click on the ws and select "Show Detectors".
- Alternatively, execute the `CreateDetectorTable` algorithm and then show data on the resulting table ws
### Platforms affected
Found on Windows, Smoke testing for release 5.1
Not a problem on 5.0

DanielMurphy22, 2021-02-11:
A very similar thing was found by @Pasarus on Ubuntu (also on Windows) for LARMOR3368 on release candidate for 6.0 https://github.com/mantidproject/mantid/issues/30655#issuecomment-776057667

martyngigg, 2021-02-23:
@peterfpeterson also mentioned on Slack that table workspaces lots of rows generally are slow. I'm going to update the description of the task to reflect that fact that the idea is to improve the speed of that here.

peterfpeterson, 2021-02-23:
The fix with `PeaksWorkspace` in mantidplot/qt4 was relatively straightforward. What is happening is the default qt-widget for rendering the table converts *everything* that may ever be rendered to a string to determine the default width for the columns; then displays the widget with only the data currently in view converted to strings. The fix for `PeaksWorkspace` was to override the method that determines the default column width to use a fixed value without inspecting the actual data. For a "generic" view, we could set a fixed value for `int`, and look at the dynamic range of `float` to return a value. This would be significantly faster than the default qt behavior.

peterfpeterson, 2021-02-23:
Finding some of the old implementation. The magic bit was [`QPeaksTableModel::numCharacters`](https://github.com/mantidproject/mantid/blob/v6.0.0/qt/widgets/sliceviewer/src/QPeaksTableModel.cpp#L103) which was used in [`PeaksWorkspaceWidget::createTableMVC`](https://github.com/mantidproject/mantid/blob/v6.0.0/qt/widgets/sliceviewer/src/PeaksWorkspaceWidget.cpp#L128-L156).

DannyHindson, 2021-02-25:
I just looked back in my notes about this issue and I think the performance used to be OK in workbench (in 5.0). And something happened in 5.1 to slow it down. At the time I started to looking into whether it was something to do with changes to qt/python/mantidqt/widgets/workspacedisplay/table/presenter.py. I was looking at whether it was this commit 0ecfe1acf089b04325d3133bc5f5d49dfc4f1572 but then ran out of time to progress it further

martyngigg, 2021-02-25:
Thanks for the pointers here, hopefully this will accelerate us in solving the issue.

PR #30791 (2021-03-04): Improve speed of table workspace display for large datasets
**Description of work.**
This PR improves the performance of table workspace display for large datasets. It does this by creating a custom table model, which loads data in batches. It should lead to speed improvements opening large tables and also improve the performance when performing column operations on the table, e.g. sorting. 

**To test:**
Load a table workspace and check everything functions as you'd expect
Load a big table workspace to observe the speed improvements, e.g. with the following script:

```
ws = Load("TOPAZ_3132_ [... cut]
closed 2021-03-04 by AndreiSavici