mantidproject/mantid #29654: Project save loading multiple projects

AnthonyLim23, 2020-10-01:
A user has said:

I’m still analysing some of my indirect data from IRIS. I’ve loaded my project that I saved and closed yesterday and today I reopened it. What’s odd is that I seem to have acquired a workspace that isn’t associated with this project, see the ringed stuff on the left hand side. Also, in my Multi_Conv_seq_1L_Parameter workspace, it’s populated it with the wrong numbers. What is going on?!

Is this a result of having more than one project saved within a folder? I’ve got two projects that I’m working on, a high temp and low temp, but the analysis will be the same (within reason).

AnthonyLim23, 2020-10-01:
@martyngigg, @Pasarus  and I have a zip of the data

martyngigg, 2020-10-05:
We think the issue here is that the project save doesn't warn users that a project is really the whole directory and saving the workspaces will overwite any with the same name.

Two options:
  - Just warn users if they will overwrite any files with workspaces from the current list
  - Create a unique filename, e.g. by appending `_i` to the filename, but recording the workspace name in the project file.

The second option is preferred but it is paramount that existing project files that only contain the filename and assume this is the workspace name will also still load.

closed 2021-09-07 by stale[bot]