mantidproject/mantid #29679: Plot context menu: Legend Show/Hide

DanielMurphy22, 2020-10-06:
**Original reporter:** Steve King

(Small feature request)

Show/hide legend on a plot can take a while (Figure Options  > Legend > Hide Legend > Apply/Ok).

Adding a `Legend > Hide/Show` option to the right-click context menu would speed up the user experience.

DanielMurphy22, 2021-06-16:
@smk78 Just thought I'd point out that something pretty similar was implemented, so that you can hide a legend by double clicking on it: https://github.com/mantidproject/mantid/pull/30751

smk78, 2021-06-16:
Oh yes! Thanks @DanielMurphy22 , I missed that feature in the release notes! You do however have to go through settings to get it back again which is not quite as convenient as this ticket would be (eg, if you are comparing different datasets on the same graph but need to keep referring to the legend). But every little helps!

PR #32112 (2021-08-09): Add option to show/hide legend in plot context menu
Adds option to show/hide legend in plot context menu so that it is quicker to do this!

**To test:**
1. Open a plot on a workspace.
2. Right click on the axes `Legend > Show/hide legend`
3. Test that this hides the legend, then test that doing it again shows the legend.

Additionally, check that when a plot is made with no legend you can still show it via `Legend > Show/hide legend`. For example on the plot generated by this script:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt

ws = CreateSampleWorkspace()
fig, ax = plt. [... cut]
closed 2021-08-09 by DannyHindson