mantidproject/mantid #29737: Fix a typo in a label on Figure Options

gemmaguest, 2020-10-15:
**Original reporter:** Max at ISIS

In a plot, go to the settings and select the Curves tab. The top item is labelled `Select an axes`. It should be `Select an axis`

joseph-torsney, 2020-11-05:
I would argue that this was fixed by pr #29137, and specifically [this commit](https://github.com/mantidproject/mantid/commit/23a60f40dd5bb5d4d19add7ca0603c52b7fbd70f), which changed _Select an axes_ to _Select a set of axes_. The combo allows the user to select which matplotlib `axes` object to modify , rather than the x or y axis for example.

martyngigg, 2020-11-23:
I can see where the initial confusion arises as matplotlib calls the the plot Axes so the original tried to be faithful to that terminology but it doesn't quite read correctly in english. I would agree that this is an improvement and I'm happy for it to be closed.
@gemmaguest Do you think that will be ok for REFL?

gemmaguest, 2020-11-23:
I'm sure that's fine. I think it's just that it read badly before that was confusing.

closed 2020-11-23 by martyngigg