mantidproject/mantid #29741: Easy way to duplicate/copy a plot's settings

gemmaguest, 2020-10-15:
**Original reporter:** Max at ISIS

In reflectometry we often have hundreds of workspaces and often want to create plots with the same look for each of them. Users will typically create one plot first and fiddle with the settings until they are happy with it. They would then like a way to be able to easily apply these settings to other plots that they create - either some way of saying that new plots should be created with the same settings as an existing plot, or a way to apply the settings from one plot to other plots. There are probably several ways that this could work so it would be worth coming up with some ideas and discussing them with Max.

DanielMurphy22, 2020-10-15:
2 thoughts on what is currently possible:
- Can generate a script from a good plot and then edit it to plot different data.
- Some of the main plotting options can be saved in File > Settings > Plots so a newly opened plot will look the way you want! Another bonus here is that the user settings can be saved to file, then shared and reloaded by someone else, essentially copying someone elses plotting settings.

I assume these don't quite go far enough for Max and it would be good to know if scripting works here? Or would saving out plot figure options (in a file or table workspace) and reapplying them be the desired solution? Is there only one "perfect set-up" that would be applied to all refl plots, or are there a few different types for which they would like to apply different sets of preferences?

gemmaguest, 2020-10-15:
@DanielMurphy22 Thanks for the suggestions. I'm not sure how much Max as experimented with this so far. I've got a meeting with him this afternoon so I'll see what he thinks.

closed 2021-04-20 by stale[bot]