mantidproject/mantid #29773: Add the Frequency Domain Analysis Interface to the PythonInterfacesStartupTest

robertapplin, 2020-10-20:
### Expected behavior
The startup script for the Frequency Domain Analysis interface (`Frequency_Domain_Analysis.py`) should be executed as part of the `PythonInterfacesStartupTest` system test.

### Actual behavior
The FDA interface is currently excluded from this system test because it crashes when you attempt to open it from this system test. This crash needs to be fixed so we can remove the `Frequency_Domain_Analysis.py` script from the excluded scripts list. **Be warned this might be a difficult/confusing issue to fix**

https://github.com/mantidproject/mantid/blob/d862e3e6045351abbfd5df4c6bfb32bd849bc723/Testing/SystemTests/tests/qt/PythonInterfacesStartupTest.py#L18

### Steps to reproduce the behavior
1. Remove the FDA interface script from the excluded list in the `PythonInterfacesStartupTest` system test.
2. Run the test. A crash will occur when trying to open the FDA interface.

### Platforms affected
Probably All, but found on windows.

martyngigg, 2021-01-06:
With the High Priority label, may I ask if this is intended to hold up the release if it weren't fixed?

robertapplin, 2021-01-06:
No I don't think this should hold up the release

PR #31469 (2021-06-14): PythonInterfacesStartupTest - close event to unsubscribe from the ADS
**Description of work.**
This PR fixes several errors and crashes when opening the Muon Analysis, Frequency Domain Analysis and Engineering Diffraction GUIs in the PythonIntefacesStartupTest. These problems were caused by not unsubscribing the interfaces from the ADS. This unsubscribing is done in the closeEvent of these interfaces, and so I need to `.close()` them to fix this issue.

**To test:**
A code review

Fixes #29773

*This does not require release notes* because **it is a system test change**

#### Reviewer ####

Pl [... cut]
closed 2021-06-14 by martyngigg