mantidproject/mantid #29855: Plot ColorFill Error when plotting a ragged workspace

robertapplin, 2020-10-29:
### Expected behavior
A Colorfill plot should plot the ragged workspace without a problem.

### Actual behavior
An error message appears in the log:
```
Error in execution of algorithm ConvertToHistogram:
ConvertToHistogram: error (see log)
Error in execution of algorithm Integration:
ConvertToHistogram: error (see log)
Traceback (most recent call last):
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\qt\applications\workbench\workbench\plugins\workspacewidget.py", line 168, in _do_plot_colorfill
    pcolormesh(names)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\plotfunctions.py", line 58, in inner_func
    return func(workspaces, *args, **kwargs)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\qt\python\mantidqt\plotting\functions.py", line 235, in pcolormesh
    pcm = pcolormesh_on_axis(ax, ws)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\qt\python\mantidqt\plotting\functions.py", line 285, in pcolormesh_on_axis
    pcm = ax.imshow(ws, cmap=ConfigService.getString("plots.images.Colormap"), aspect='auto', origin='lower',
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\mantidaxes.py", line 46, in wrapper
    func_value = func(self, *args, **kwargs)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\mantidaxes.py", line 849, in imshow
    return self._plot_2d_func('imshow', *args, **kwargs)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\mantidaxes.py", line 875, in _plot_2d_func
    plotfunctions_func(self, *args, **kwargs),
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\axesfunctions.py", line 644, in imshow
    return samplingimage.imshow_sampling(axes, workspace=workspace, *args, **kwargs)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\resampling_image\samplingimage.py", line 227, in imshow_sampling
    im._resample_image(100, 100)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\resampling_image\samplingimage.py", line 108, in _resample_image
    x, y, data = get_matrix_2d_ragged(self.ws,
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\datafunctions.py", line 517, in get_matrix_2d_ragged
    counts = interpolate_y_data(workspace, x_centers, y, normalize_by_bin_width, spectrum_info=spec_info,
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\datafunctions.py", line 567, in interpolate_y_data
    workspace_indices = _workspace_indices_maxpooling(y, workspace) \
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\plots\datafunctions.py", line 552, in _workspace_indices_maxpooling
    summed_spectra_workspace = Integration(workspace)
  File "c:\users\mlc47243\documents\mantid-development\mantid-2\mantid\framework\pythoninterface\mantid\simpleapi.py", line 1036, in __call__
    algm.execute()
RuntimeError: ConvertToHistogram: error (see log)
ConvertToHistogram: Histogram: size mismatch of data.
```

### Steps to reproduce the behavior
1. Run this script to create a ragged workspace
```
ws = CreateWorkspace([0,1,2,3,4,2, 3,4,5,6],[0,1,2,3,4,3,4,5,6,7], NSpec=2)
min = 1
max = 6
for j in range(0,2):
    x =[]
    y =[]
    data = ws.readY(j)
    k=0
    for v in ws.readX(j):
        if v >= min and v<=max:
            x.append(v)
            y.append(data[k])
        k+=1
    tmp = CreateWorkspace(x,y)
    ConjoinWorkspaces(ws, tmp, CheckOverlapping=False)
```
2. Right click the workspace and select `Plot ColorFill`
3. The error message appears.

### Platforms affected
Found on windows, probably All. This bug was not present in Mantid Version 5.1.0

StephenSmith25, 2020-10-29:
This seems to be an issue with something I added known as maxpooling, which ensures that the high counting spectra are represented when an image is downsampled to the number of pixels in the figure. For ragged workspaces this should probably be ignore, maybe a check can be added here : https://github.com/mantidproject/mantid/blob/4eb6d899a9ac9df16ee46574e3edc139f940e081/Framework/PythonInterface/mantid/plots/resampling_image/samplingimage.py#L145

robertapplin, 2020-10-29:
> This seems to be an issue with something I added known as maxpooling, which ensures that the high counting spectra are represented when an image is downsampled to the number of pixels in the figure. For ragged workspaces this should probably be ignored, maybe a check can be added here :
> 
> https://github.com/mantidproject/mantid/blob/4eb6d899a9ac9df16ee46574e3edc139f940e081/Framework/PythonInterface/mantid/plots/resampling_image/samplingimage.py#L145

Thanks, I'll look into adding this check

PR #29862 (2020-11-18): Fix Colorfill Plotting for Ragged Workspaces
~~Merge after #29859~~ 
---------------------

**Description of work.**
This PR fixes a bug where attempting to Plot Colorfill on a ragged workspace causes an error.

**To test:**
1. Run this script to create a ragged workspace
```
ws = CreateWorkspace([0,1,2,3,4,2, 3,4,5,6],[0,1,2,3,4,3,4,5,6,7], NSpec=2)
min = 1
max = 6
for j in range(0,2):
    x =[]
    y =[]
    data = ws.readY(j)
    k=0
    for v in ws.readX(j):
        if v >= min and v<=max:
            x.append(v)
            y.append(data[k])
        k+=1
    tmp = CreateWorkspace(x,y)
    [... cut]
closed 2020-11-18 by gemmaguest