mantidproject/mantid #29886: Indirect Data Analysis: Handle WorkspaceIndex variable for Q dependent functions

StephenSmith25, 2020-11-03:
In Q dependent functions either a `Q` value or `WorkspaceIndex` is needed. In the indirect GUI a Q value is specified, which leaves the `WorkspaceIndex `at its default value, which is taken as the maximum size an integer can be. This leaves a large number (_circa_ 2147483647) in the function browser, which looks like an error. This issue is to deal with this case in a better and less confusing (to the user) way. E.g by either defaulting it as -1, or graying out `WorkspaceIndex` attribute box.

### Steps to reproduce the behavior
- Open the Indirect data analysis GUI
- Go to ConvFit and load some data
- Add a Q dependent function (TeixeiraWater) click show full function view and you'll see a large integer in the WorkspaceIndex box.

### Platforms affected
All

AnthonyLim23, 2020-12-07:
@Harrietbrown is this now fixed?

StephenSmith25, 2020-12-07:
I'm not surr if this is fixed, as it was a new issue exposed when fixing attributes in the interface. It links back to the `FunctionQDepends`, which defaults the `WorkspaceIndex` to be an `EMPTY_INT`. As we don't overwrite this parameter in the interface, it remains an `EMPTY_INT` througout, which is an alias for `int_max`. Changing it to -1 would be a simple solution, but Spencer had suggested that a better solution would be to gray out the box, however this may be slightly more complicated.

Harrietbrown, 2020-12-08:
It is still around, I'm going to add this to the icebox untill after the refactor since it doesn;t interfear with the functionality of the interface.

SpencerHowells, 2021-12-05:
@Harrietbrown please keep open

Harrietbrown, 2022-12-13:
Not stale

Harrietbrown, 2023-01-24:
[image] [Indirect Data Analysis: Handle WorkspaceIndex variable for Q dependent functions](https://trello.com/c/Aexg1UUG/14-indirect-data-analysis-handle-workspaceindex-variable-for-q-dependent-functions)

AnthonyLim23, 2023-07-31:
This seems to be fixed

AnthonyLim23, 2023-07-31:
I looked this morning and it seemed to be ok

PR #36553 (2023-12-14): Disable unused WorkspaceIndex attribute for Q dependent functions
### Description of work
This PR ensures the `WorkspaceIndex` int property in the FunctionTreeView is disabled when it is not being used. When the property is not being used, it is defaulted to the maximum numeric limit for an int. This avoids confusing users who see an unexpectedly large WorkspaceIndex attribute in their function.

Fixes #29886

### To test:
1. Open the Inelastic data analysis GUI
2. Go to ConvFit 
3. Select a Q dependent function (TeixeiraWater)
4. click show full function 
5. The large WorkspaceIndex integer p [... cut]