mantidproject/mantid #29892: Muon fitting refactor

AnthonyLim23, 2020-11-03:
The muon fitting is very complex and needs simplifying.

AnthonyLim23, 2021-01-12:
As a base line:

The fitting tab will have an option to select the "mode" (we will only consider fitting here, TF asymm will be done in later issue)
The TF-Asymmetry option will be removed from the table
The code will be simplified and stripped back in a way that can be easily overloaded.
It will work with sequential fitting
If double pulse is selected on the home tab, it will do a double pulse fit (I will add an example)

AnthonyLim23, 2021-01-12:
I’ve had a chance to perform some measurements on boron using ARGUS in both single pulse and double pulse so we can compare the fitting of double pulse and single pulse data in Mantid. The details are below:

Double Pulse
•	71796 – TF20
•	71797 – ZF (50 MEv)
•	71798 – LF20 (25 MEv)

Single Pulse
•	71799 – TF20
•	71800 – ZF (50 MEv)
•	71801 – LF20 (15 MEv – I ran out of time)

robertapplin, 2021-02-04:
This is a diagram of what the new file structure will look like for the fitting tab.

[image]

AnthonyLim23, 2021-02-04:
looks good. We will also need to add the appropriate contexts. You can probably just create multiple (using inheritance again) fitting contexts. 

The context contains the information from the tab. In theory we should be able to populate the whole GUI by just looking at the contexts.

robertapplin, 2021-02-05:
File structure diagram 2.0

[image]

robertapplin, 2021-02-05:
[image]

AnthonyLim23, 2021-02-15:
For single pulse:
1.	Set time zero to 0.493
2.     Load Argus 71799
3.	Calculate alpha
4.	Load the next run
5.	Do a fit with a StaticKuboToyabeTimeExpDecay and FlatBackground
6.	A = 0.12, Delta = 0.56, Lambda = 0.12 and A0 = 0.09

For double pulse (set on home tab of muon analysis):
1.	Set time zero to `from data file`
2.     Load Argus 71796
3.	Calculate alpha (it is slightly different to single pulse)
4.	Load the next run
5.	Do a fit with a StaticKuboToyabeTimeExpDecay and FlatBackground
6.	A = 0.12, Delta = 0.55, Lambda = 0.13, A0 = 0.09

robertapplin, 2021-02-26:
Final file structure diagram
[image]

dotted arrow means 1 `has a` 2
full arrow means 1 `inherits from` 2

robertapplin, 2021-02-26:
I will probably create a design doc or something to detail the changes made to this fitting stuff

AnthonyLim23, 2021-03-02:
in the TFAsymmetry widget can we have a tick box to fix the value?

robertapplin, 2021-03-02:
Fix the value of what? I think it would be best to do that separate to the refactor

AnthonyLim23, 2021-03-02:
> Fix the value of what? I think it would be best to do that separate to the refactor

ah sorry. A tick box to say we want to fix the normalisation to its current value

PR #30806 (2021-03-31): Muon Fitting Refactor 3 - Make Sequential Fitting tab work
Follows on from #30772
---------------------------
**Description of work.**
This PR ensures the sequential fitting tab works with the refactored fitting widgets in the fitting tab.

It also:
- [x] Updates tests for `TFAsymmetryFittingModel`
- [x] Fixes and re-enable `SeqFittingTabPresenterTest`

**To test:**

Fixes #29892

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the co [... cut]
PR #30772 (2021-04-07): Muon Fitting Refactor 2 - Add TF Asymmetry fitting
Follows on from #30708
--------------------------
**Description of work.**
This PR continues with the refactor for Muon Analysis. It implements the TF Asymmetry fitting mode.

Tests have been added for:
- [x] `TFAsymmetryFittingView` test
- [x] `TFAsymmetryFittingPresenter` test
- [x] `TFAsymmetryFittingModel` test
- [x] `TFAsymmetryFittingOptionsView` test

Things to do in a future PR:
- Get the sequential fitting tab to work with the refactored code

**To test:**
Try out the fitting in the fitting tab of Muon Analysis and FDA. Try:
- Single  [... cut]
PR #30708 (2021-04-22): Muon Fitting Refactor
**Description of work.**
This PR is the base branch for the refactor of the muon fitting widgets. It creates the basic and general fitting widgets which can be used for normal fitting in the fitting tab of Muon Analysis and FDA interfaces. It also creates a widget which can be used for TFAsymmetry fitting.

Tests have been added for:
- [x] Basic model, view and presenter
- [x] General model, view and presenter
- [x] TFA model, view and presenter
- [x] `FittingTabPresenterTest` and `FittingTabModelTest` have been removed.

**To test:**

Fixes #29892

#### Reviewer ####

Pl [... cut]
closed 2021-04-22 by martyngigg