mantidproject/mantid #29899: Allow PaalmanPingsMonteCarloAbsorption to run on workspace with sample shape already defined

DannyHindson, 2020-11-05:
The PaalmanPingsMonteCarloAbsorption algorithm only supports a mode of operation where the algorithm creates the sample shape based on the shape specified in the algorithm parameters (which is either Flat Plate, Cylinder or Annulus). This issue covers work to extend the algorithm so it can also run on a sample shape that is already defined in the input workspace. This will extend its range of use to cover mesh sample shapes and it will also align it better with the new PaalmanPingsAbsorptionCorrection algorithm

@gvardany - this is the task we discussed in the call with the ORNL guys a couple of weeks ago. May get chance to do this in our next sprint which will be ~Dec

gvardany, 2020-11-05:
Great, we should make sure to allow any combination of predefined sample/container shape/material. This will allow not only mesh but also custom CSG geometries.

The main issue is to ensure that when doing container only, sample should not scatter, when doing sample only, the container shouldn't scatter. Back then I was unsure wether one can have a container without a sample. If it turns out impossible, the trick would be to set either the cross-section to zero, or the height (so the volume) to zero respectively.

DannyHindson, 2020-11-13:
@gvardany - I had a call with one of the indirect instrument scientists at ISIS yesterday and (coincidentally) they would like me to support mesh\CSG geometries in PaalmanPingsMonteCarloAbsorption. So the ISIS requirement to do this is a bit more solid than it was previously. So I will probably start this work shortly.
I think I should remove the container input workspace so the algorithm accepts a single workspace which is simply called "Input Workspace". Otherwise it's confusing for the user to know which input workspace to attach the complex shapes to. This would be a "breaking" change but the algorithm was only added recently so it feels like we could make an exception here. Would you be happy for me to do this? The ISIS users of this algorithm use it via the Indirect, Corrections user interface so as long as I change that also they will be OK.

gvardany, 2020-11-13:
Indeed, for the predefined shapes, it better be a single workspace. It is a minimally breaking change, so I would say it is acceptable.

PR #30159 (2021-02-02): Allow PaalmanPingsMonteCarloAbsorption to use shape defined in workspace
**Description of work.**

Previously PPMCA would require the user to define a new sample (and possibly container) shape before running, however there is now the option to use the shape preset in the workspace data.
If the preset shape is being used, there is still the option to override the sample material without affecting the sample and can geometries. 
As part of this, the algorithm now only takes one input workspace for both sample and container rather than one for each.
The Indirect Corrections Calculate Monte Carlo  [... cut]
closed 2021-02-02 by gvardany