mantidproject/mantid #29900: Colorfill plot sometimes produces

martyngigg, 2020-11-05:
### Expected behavior

A colorfill plot in workbench should not produce any other data

### Actual behavior

A separate workspace is generated along with logging that integration was run

### Steps to reproduce the behavior

Load any smaller 2D workspace and plot a colorfill plot.

### Platforms affected
All

martyngigg, 2020-11-05:
@StephenSmith25 I guess this is because of the changes we made to use the maxpooling?
Would you be able to take a look at this as a priorty over the window on top stuff?

StephenSmith25, 2020-11-05:
Yeah, my bad - didn't spot it was spitting out a workspace until a couple of days ago 🤦‍♂️ . I had already added a `StoreInAds=False` flag in the colorbar scale PR, but is there a way I can suppress the Integration message, if so I'll create a PR for it shortly.

martyngigg, 2020-11-05:
No problem. I should have caught that in the review :)
All of the functions have an `EnableLogging` keyword so you can just set `EnableLogging=False`.

Quick question: Is it expected that `Integration` is run twice?

StephenSmith25, 2020-11-05:
I'm thinking this only occurs when you first open the colorfill (or sliceviewer) as we open it with a default size and then rescale it based on the screen size (/resolution). If you zoom in you'll notice it only integrates once when doing the resampling.

PR #29910 (2020-11-11): Fix colorfill and sliceviewer unnecessary logging and system exception
**Description of work.**
This PR fixes the unnecessary logging produced by the maxpooling stage performed when sliceviewer and colorfill images are resampled. It also fixes another issue which causes a system exception due to `integration` running as a managed algorithm.

**To test:**
- Load some data
- Open sliceviwer and colorfill and make sure everything opens correctly
- Zoom in a resize the window to test that resampling still works.

Fixes #29900 . 

This does not require release notes because they were issues introdu [... cut]
closed 2020-11-11 by martyngigg