mantidproject/mantid #29912: Fit Wizard - Design/layout ideas

robertapplin, 2020-11-06:
### Description
A basic design/layout for the Fit wizard interface should be thought-up/drawn before work begins on its creation.

Any ideas/designs can be added as a comment in this issue.

robertapplin, 2020-11-10:
[image]

robertapplin, 2020-11-10:
@AnthonyLim23 What do you think about this initial design. Anything that needs changing?

AnthonyLim23, 2020-11-16:
it looks good. When we meet the scientists (should be soon), I would like you to present it. They can then give feedback.

robertapplin, 2020-12-03:
Closed because this design was agreed. The design has been implemented in #30068 so this issue is no longer required. Future changes to the design can be raised in separate issues

closed 2020-12-03 by robertapplin