mantidproject/mantid #29947: RuntimeError in figuremanager when renaming a workspace

joseph-torsney, 2020-11-12:
### Expected behavior
With a plot open, after renaming the workspace for the plot, the window and axes titles should update to match the new name.

### Actual behavior
The workspace is indeed renamed, but the plot is not updated and the following runtime error occurs
```
Error occurred in handler:
Traceback (most recent call last):
  File "C:\Users\ggk63474\programming\mantid\qt\applications\workbench\workbench\plotting\figuremanager.py", line 51, in wrapper
    func(*args, **kwargs)
  File "C:\Users\ggk63474\programming\mantid\qt\applications\workbench\workbench\plotting\figuremanager.py", line 143, in renameHandle
    for ws_name, artists in ax.tracked_workspaces.items():
RuntimeError: dictionary keys changed during iteration
```

### Steps to reproduce the behavior
1. Load a workspace
2. Plot a spectrum
3. Run `rename` on the workspace

### Platforms affected
Found on Windows

Whilst working on #29258, I am seeing that when `rename` runs, both the `SliceViewerADSObserver`s `renameHandle` and `replaceHandle` are fired. Maybe something similar is happening with the figuremanager's observer, and the `replaceHandle` is changing that dictionary?

joseph-torsney, 2020-11-12:
Remote debugger connected to both `replaceHandle` and `renameHandle` of `FigureManagerADSObserver`!

samtygier-stfc, 2020-11-30:
I can reproduce on Windows but not Centos7

PR #30493 (2021-02-03): Iterate over list of dict items to prevent runtime error when renaming plotted workspace
**Description of work.**

Small bug fix - Iterate over list of dict keys to prevent runtime error when renaming plotted workspace - this avoids a runtime error that arose due to change to the underlying dict of the items iterator.

**To test:**

(1) Plot spectrum
(2) Rename workspace

The legend and/or title (depending on the plot) won't change - but you can tell the tracked workspaces has been update because if you delete the workspace the ADS observer should lead to the plot being closed.

Partially Fixe [... cut]
PR #30723 (2021-03-01): Update curve labels when a workspace is renamed
**Description of work.**
This PR implements a method to re-label a plot when a workspace is renamed in the ADS. 

**To test:**
- Load some data
- Make a 1d plot
- Rename the workspace
- Things should update on the plot (legend, title)
- Try this multiple times for different plots/ different data. It should work

Fixes #29947.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code [... cut]
closed 2021-03-01 by martyngigg