mantidproject/mantid #29957: Move docs-qthelp target back into Documentation folder in Visual Studio

DannyHindson, 2020-11-13:
As part of the recent work to decouple the docs build from MantidPlot, the docs-qthelp target has been moved out of the Documentation folder in Visual Studio. I discussed this with Martyn and I don't think that was intentional. This issue covers moving it back into that folder. 

@martyngigg - I've created this issue as another small tidy up on the docs. Out of interest why was the original docs-qthelp target split into docs-qtassistant and docs-qthelp? One seems to compile the individual html pages and the other seems to create .qhc\.qch files. Maybe people using Qt Creator can view the output of docs-qtassistant directly without needing to build the docs-qthelp part?

martyngigg, 2020-11-16:
@DannyHindson Thanks for creating this. The original code was always a two step process but the first Sphinx part was run as custom command rather than a custom target. When I cleaned that cmake code up to reduce the duplication it was simpler to also create a target for running Sphinx using the same cmake function the others used. My aim was for devs to still just really use old `docs-qthelp` target that should just do the same thing. CMake targets tend to be a little easier to work with too especially when setting dependencies. Happy to switch things around if that helps.

DannyHindson, 2020-11-16:
@martyngigg Thanks for the clarifications. We might want to change the docs-qtassistant target back into a custom command for a couple of reasons to help Visual Studio users:
a) If we can't get the dependency to work in Visual Studio between docs-qthelp and docs-qtassistant. This is something to do with dependencies not working nicely with the EXCLUDE_FROM_ALL and EXCLUDE_FROM_DEFAULT_BUILD settings but haven't bottomed this out yet
b) Whether the docs-qtassistant target causes too much of a distraction given that VS users are building targets by looking at the solution explorer where they see all the target names. Perhaps linux users building targets by typing "ninja docs-qthelp" from the command line are less distracted by a new target

I've just created a PR to move the docs-qthelp target back into Documentation folder for now

martyngigg, 2020-11-16:
@DannyHindson Thanks for investigating this. I completely agree. I think the main thing is that the `docs-qthelp` is in the right place for users. The command line builders do seem to get the dependencies correct so that typing `docs-qthelp` works but that's no good if it doesn't work everywhere. The qtassistant step is just a detail so not really important. If it could be worked out as a target then maybe we could put it in a subfolder named "detail" or something? If not then the custom command option seems like the sensible approach.

PR #29961 (2020-11-17): Add folder to the docs-qthelp target (only affects VS)
**Description of work.**

Move the docs-qthelp target back into the Documentation folder (only affects Visual Studio)

Also fix problem with import readline that was preventing docs build on Windows. I think this module should only be imported on Linux. Have wrapped it in an "if sys.platform" statement that I saw used in qt\applications\workbench\workbench\app\start.py

**To test:**

Run cmake on Windows and check the target shows up in the Documentation folder in Visual Studio:

[image]

Build the docs-qtassistant target and then the docs [... cut]
closed 2020-11-17 by martyngigg